CongducPham/LowCostLoRaGw #156: Error Adafruit Feather 32u4.

miryamparedes, 2017-11-27:
Hello. I am using an Adafruit Feather 32u4 and when trying to compile the Arduino_Lora_Ping_Pong I get this message error:
exit status 1
Error compiling for board Adafruit Feather 32u4.

Please could you let me know what could be the problem? Thank you.

CongducPham, 2017-11-27:
Well, I don't have this issue. What is the version of the adafruit lib you have? I have the latest one, 1.4.9, and it compiles without error.

miryamparedes, 2017-11-28:
I am using that version as well. I have no idea what is wrong. Do you have any idea? Thank you. 

[image]

CongducPham, 2017-11-28:
Maybe you are missing a lib. Please give more details on the error message.

ethicstechOSS, 2017-11-29:
@miryamparades Most probably you forget to install the SX1272 library on your arduino IDE. Just installing the Adafruit package is not enough to compile a LoRa program successfully.

miryamparedes, 2017-11-29:
Dear all, thank you for your help. I made a mistake when installing the libraries in the wrong file.

closed 2019-09-13 by miryamparedes