CongducPham/LowCostLoRaGw #164: "Packet sent, state 1" with sx1278

yongxiangu, 2018-01-23:
The gateway can receive the data from endnode. The endnode prints "Packet sent, state 1". 

After enable DEBUG in sx1272.h, I find "Packet sent, state 1" problem is from "int8_t SX1272::setPacketLength(uint8_t l)", line "if( packet_sent.length == value )". 

packet_sent.length is always 0xFF, while value is not 0xFF. So setPacketLength(uint8_t l) always returns 1.

CongducPham, 2018-01-23:
What example are you using?

yongxiangu, 2018-01-23:
Thank you for your reply! The example is Arduino_LoRa_Ping_Pong

manelmehernaz, 2018-05-31:
hi there! did u solve the issue ? 
am having the same problem , after compiling the Lora_ping_pong sketch, i get in the serial monitor "packet sent , state 1"
am using two arduino with 2 inair9B
on the arduino_lora_gateway side i get the result in he following pic 
could u @CongducPham explain what is in there ? 
thanks!
[image]

CongducPham, 2018-06-01:
Seems that the gateway receives the Ping. Maybe a current issue, that sometime happens, the inAir9B consumes more power than the inAir9.

manelmehernaz, 2018-06-01:
Thanks for your reply ! is there a way to fix it ?