CongducPham/LowCostLoRaGw #166: Changes made in lora gateway files using terminal is not reflecting on browser UI

Ram-Murali, 2018-02-03:
Hi, 
    I'm using Raspberry pi Low cost lora gateway.On terminal my gateway is working good.But the problem is whatever changes made in gateway files is not reflecting on browser UI.
     I'm currently working on customizing gateway UI with the help of files in link shown below:
https://github.com/CongducPham/LowCostLoRaGw/tree/master/gw_full_latest/gw_web_admin/admin

Any help would be appreciated
Thanks in Advance

CongducPham, 2018-02-04:
That strange. Try install the latest version of the web interface. If already done, then I don't actually see why changes are not reflecting on the web interface. I've tested and it works for me.

Ram-Murali, 2018-02-05:
thanks for your support.

closed 2018-02-05 by Ram-Murali