CongducPham/LowCostLoRaGw #168: RadioHead lib and lora_gateway

d00m178, 2018-02-08:
Hello, my question is regarding RadioHead libs.
http://www.airspayce.com/mikem/arduino/RadioHead/index.html

I already saw the similar topic here:
https://github.com/CongducPham/LowCostLoRaGw/issues/141
and example for it.

But in my case now I can't use the variant with changing source code for node side because all my nodes already installed outdoor, and I need to wait at least 2-3 month for get it back to re-flash new firmware.
So I'm looking the way to change the source code on gateway side library.

Is it possible to change lora_gateway packet header to correspond it to header in RadioHead lib?
Gateway's header is: dst(1B) ptype(1B) src(1B) seq(1B) 
RadioHead's header is: to, from, id, flags    

to <-> dst ; from <-> ptype ; id <-> src ; flags <-> seq 

so I suppose I need change Gateway's header to: dst(1B) src(1B) ptype(1B) seq(1B) 
so correspondence will be:
to <-> dst ; from <-> src ; id <-> ptype ; flags <-> seq 
and I won't change the code on my nodes side.

Is it right guess? If yes - in which file I need search this header and maybe I will need change also some another code?

CongducPham, 2018-02-08:
Set the gateway in raw format mode and then change some lines in post_processing_gw.py to decode and accept your RadioHead packet.

CongducPham, 2018-02-08:
Change lines where you see the comment:
#dissect our modified Libelium

d00m178, 2018-02-20:
Thank you, but in RAW format (--raw key) gateway says:

Raw format, not assuming any header in reception

and then prints zero in source and destination:
--- rxlora. dst=0 type=0x00 src=0 seq=0 len=11 SNR=6 RSSIpkt=-46 BW=125 CR=4/5 SF=7

so I don't understand what is the point to change header in post_processing_gw.py
will it change something?

CongducPham, 2018-02-21:
When the gateway runs in raw format, the low-level gateway is not decoding the header, so in a first step, there is no information on the packet passed by the low-level gateway. But, if you look at line 1037 of post_processing_gw.py "if (_rawFormat==1)" then the post-processing stage can decode the header and fill in the required information. This is what you can do. regards,

d00m178, 2018-02-21:
thank you. 
I changed code in post_processing_gw.py:

```
....................
                                if ord(ch)==1:
                                        dst=ord(ch)
                                        ###change order
                                        src=ord(getSingleChar())
                                        ###
                                        ptype=ord(getSingleChar())
                                        ###src=ord(getSingleChar())
                                        seq=ord(getSingleChar())
                                        ###print "Header[dst=%d ptype=0x%.2X src=%d seq=%d]" % (dst,ptype,src,seq)
                                        print "Header[dst=%d src=%d ptype=0x%.2X seq=%d]" % (dst,src,ptype,seq)
                                        #now we read datalen-4 (the header length) bytes in our line buffer
                                        fillLinebuf(datalen-HEADER_SIZE)
                                        datalen=datalen-HEADER_SIZE
                                        ###pdata="%d,%d,%d,%d,%d,%d,%d" % (dst,ptype,src,seq,datalen,SNR,RSSI)
                                        pdata="%d,%d,%d,%d,%d,%d,%d" % (dst,src,ptype,seq,datalen,SNR,RSSI)
                                        print "update ctrl pkt info (^p): "+pdata

....................

```
Then I had to set  "raw" : true, in gateway_conf.json
and run gateway with such command:
sudo ./lora_gateway --raw --bw 125 --cr 45 --sf 7 --freq 915 | python ./post_processing_gw.py
 
and after this I received right src tag in gateway output:

```
got first framing byte
--> got LoRa data prefix
raw format from LoRa gateway
Header[dst=1 src=5 ptype=0x20 seq=0]
update ctrl pkt info (^p): 1,5,32,0,7,6,-72
Hello!
--- rxlora. dst=0 type=0x00 src=0 seq=0 len=11 SNR=6 RSSIpkt=-71 BW=125 CR=4/5 SF=7
2018-02-21T20:20:40.759950
rcv ctrl pkt info (^p): 0,0,0,0,11,6,-71
splitted in:  [0, 0, 0, 0, 11, 6, -71]
rawFormat(len=11 SNR=6 RSSI=-71)
rcv ctrl radio info (^r): 125,5,7
splitted in:  [125, 5, 7]
(BW=125 CR=5 SF=7)
rcv timestamp (^t): 2018-02-21T20:20:40.758

```

but after while gateway output starts to show this:

```
got first framing byte
--> got LoRa data prefix
raw format from LoRa gateway
Header[dst=1 src=5 ptype=0x32 seq=0]
update ctrl pkt info (^p): 1,5,50,0,7,6,-72
--> DATA with_appkey: read app key sequence
app key is  0x48 0x65 0x6c 0x6c
app key disabled
o!
--- rxlora. dst=0 type=0x00 src=0 seq=0 len=11 SNR=7 RSSIpkt=-72 BW=125 CR=4/5 SF=7
2018-02-21T20:22:35.744565
rcv ctrl pkt info (^p): 0,0,0,0,11,7,-72
splitted in:  [0, 0, 0, 0, 11, 7, -72]
rawFormat(len=11 SNR=7 RSSI=-72)
rcv ctrl radio info (^r): 125,5,7
splitted in:  [125, 5, 7]
(BW=125 CR=5 SF=7)
rcv timestamp (^t): 2018-02-21T20:22:35.743

```
please note - o! instead Hello!

and moreover after this it starts shows:

```
got first framing byte
--> got LoRa data prefix
raw format from LoRa gateway
Header[dst=1 src=5 ptype=0x3F seq=0]
update ctrl pkt info (^p): 1,5,63,0,7,6,-72
--> DATA encrypted: encrypted payload size is 7
--> DATA encrypted: local aes not activated
--> FYI base64 of LoRaWAN frame w/MIC: SGVsbG8hAA==
--> number of enabled clouds is 0
--> discard encrypted data

--- rxlora. dst=0 type=0x00 src=0 seq=0 len=11 SNR=6 RSSIpkt=-71 BW=125 CR=4/5 SF=7
2018-02-21T20:24:04.968614
rcv ctrl pkt info (^p): 0,0,0,0,11,6,-71
splitted in:  [0, 0, 0, 0, 11, 6, -71]
rawFormat(len=11 SNR=6 RSSI=-71)
rcv ctrl radio info (^r): 125,5,7
splitted in:  [125, 5, 7]
(BW=125 CR=5 SF=7)
rcv timestamp (^t): 2018-02-21T20:24:04.961
```

it's not clear why it shows that DATA encrypted?
I didn't encrypt at all..
for send data I use simple code from RadioHead lib:

```
.................
uint8_t data[] = "Hello!";

...........
 if (manager.sendtoWait(data, sizeof(data), srv)) // STEP_1_start: send data request (>) to server_number and waits 600ms (default: 3 retries * 200 timeout ms) for ACK.

```

and in gateway output I receive messages with "Hello!", then messages with "o!" and then with DATA encrypted. 
and then again repeats.

CongducPham, 2018-02-22:
Well, if you want to get a packet that has been sent with the RadioHead lib then you have to decode the RadioHead header which is: to, from, id, flags. Then change the post_processing_gw.py section accordingly, you don't have to stick with our header as some fields do not have RadioHead correspondance, e.g. ptype, src, seq for instance.