CongducPham/LowCostLoRaGw #172: Can't communicate between Arduino_Simple_Temp <-> Arduino_LoRa_Gateway

cassiolucas, 2018-02-22:
Dear Mr. @CongducPham,

I am trying to test the code using two arduinos and two NiceRF1276 V2.0 (915MHz) modules, so one is the Simple_Temp and the other is Gateway (Arduino_LoRa_Simple_temp and Arduino_LoRa_Gateway).

The only parameter I have changed on both is the #define BAND900

The Simple Temp looks fine, because it loads the module and start sending messages:
 
[image]

But I dont receive anything on the gateway side:

[image]

Do you know why? Maybe there is another parameter which I need to change?

CongducPham, 2018-02-22:
Check the PABOOST in the Arduino_LoRa_temp as the NiceRF needs PABOOST. Do the same for the gateway.

cassiolucas, 2018-02-22:
I have already tried this but no success, anyway I will give it one more
try... Any other idea?

Thank you.

cassiolucas, 2018-02-22:
Hello @CongducPham 

Yes, I have enabled the PABOOST and now it works.

I am just wondering why the RSSI value is not close to 0, the gateway and sensor are just side-by-side:

[image]

CongducPham, 2018-02-23:
When nodes are close, the value you got is normal. You will never get something close to 0.