CongducPham/LowCostLoRaGw #173: downlink questions

d00m178, 2018-02-22:
I want to send requests to my node via gateway downlink feature.

Node sleep 60 seconds, then wake up, listen for request and if it hear its own number ID in this request - node starts to measure sensors and then respond the data to the destination address which sent request.
It is implemented with help of RadioHead library.
So I want to gateway send this request and receive answer from node.
Is it possible with using downlink?

I've read this manual:
https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-downlink.md
and if I understand all correct - it can sent downlink message to node only after receiving some data from node.
But I want gateway first send request to node without previous receiving data from this node.   

Another questions - is it necessary to build lora_gateway program like stated here:
https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-downlink.md#build-on-your-raspberry

**make lora_gateway_pi2_downlink**

Because in both cases - 
**make lora_gateway_pi2**
and
**make lora_gateway_pi2_downlink**

my gateway doesn't send requests to node..
output is:

```
Running interactively the full LoRa gateway... CTRL-C to exit
sudo ./lora_gateway --raw --bw 125 --cr 45 --sf 7 --freq 915 | python post_processing_gw.py | python log_gw.py
raw output from low-level gateway. post_processing_gw will handle packet format
.................
Current working directory: /home/pi/LowCostLoRaGw/gw_full_latest
SX1276 detected, starting.
SX1276 LF/HF calibration
...
**********Power ON: state 0
Default sync word: 0x12
LoRa CR 45: state 1
LoRa SF 7: state 0
LoRa BW 125: state 0
Frequency 915.000000: state 0
Set LoRa power dBm to 14
Power: state 0
Get Preamble Length: state 0
Preamble Length: 8
LoRa addr 1: state 0
Raw format, not assuming any header in reception
SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
Low-level gw status ON
2018-02-22 19:23:25.422168
post downlink: checking for downlink/downlink-post.txt
post downlink: no downlink requests
post downlink: list of pending downlink requests
{"status":"send_request","dst":2,"data":"2"}
2018-02-22 19:23:30.427543
post downlink: checking for downlink/downlink-post.txt
post downlink: no downlink requests
post downlink: list of pending downlink requests
{"status":"send_request","dst":2,"data":"2"}
2018-02-22 19:23:35.432870
post downlink: checking for downlink/downlink-post.txt
post downlink: no downlink requests
post downlink: list of pending downlink requests
{"status":"send_request","dst":2,"data":"2"}
2018-02-22 19:23:40.438200
post downlink: checking for downlink/downlink-post.txt
post downlink: no downlink requests
post downlink: list of pending downlink requests
{"status":"send_request","dst":2,"data":"2"}
```

and in lora_gateway/downlink/ directory only downlink-post-queued.txt file and no file downlink.txt

Also during reading this document, seems I found some typos:
for example, here:
https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-downlink.md#use-scp

echo create filename **downlink-post.txt**, but scp uses filename **downlink.txt**
the same here with curl:
https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-downlink.md#use-an-http-server-with-upload

I'm not sure if it is typos but looks like it is, so this is misleading..

d00m178, 2018-02-23:
UPD about my first question regarding downlink to RFM95+RadioHead node.
I have search for another "downlink questions" here and noticed information about Lora Words:
for example - https://github.com/CongducPham/LowCostLoRaGw/issues/93#issuecomment-302071118

Then I find out that in gateway source code Lora Words is in **SX1272.h**:
```
// added by C. Pham
#define        REG_SYNC_WORD                    0x39
```
In RadioHead libs I found mention about Lora Words only in **RH_RF95.cpp** :

```
    // Set up default configuration
    // No Sync Words in LORA mode.
    setModemConfig(Bw125Cr45Sf128); // Radio default

```
so I wonder what I need to change in **SX1272.h** REG_SYNC_WORD? Maybe just 0?

Also I turn on PA_BOOST on gateway config because I'm using RFM95 module with my RPI.

d00m178, 2018-02-23:
I just noticed that there is info - **Default sync word: 0x12** during starting lora_gateway:

```
SX1276 detected, starting.
SX1276 LF/HF calibration
...
**********Power ON: state 0
Default sync word: 0x12

```
so I supposed that everything fine with this "Lora Sync Word"

So why lora_gateway doesn't send downlink requests?

CongducPham, 2018-02-27:
Yes, 0x12 is for non-loraWAN networks. For the gateway to send downlink, you first have to compile with downlink feature:

make lora_gateway_downlink
or
make lora_gateway_pi2_downlink

depending on your raspberry.

Then, in gateway_conf.json, you have to set the downlink interval to a value greater than 0, e.g. 300s for every 5 min. Please have a look at the dedicated readme. https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-downlink.md

CongducPham, 2018-02-27:
I guess your problem is that the gateway waits for a message from node 2 in order to send the downlink request.

d00m178, 2018-02-27:
Hi, 
thanks, but I've already read all document README-downlink.md.
I want gateway send request to node without previous receiving message from this node.
Is it possible to do that?

CongducPham, 2018-02-27:
You have to change many things in the lora_gateway.cpp program to change the behavior so that the program checks for downlink.txt periodically and not only on packet reception. Then you have to generate directly the downlink.txt file instead of relying on post_processing_gw.py.

d00m178, 2018-02-27:
thank you. so it doesn't implemented in current version of lora_gateway.
ok, need to check if I can change it.

plardin001, 2018-03-25:
Hi ! 
I'm trying to do the same and will be very interested if you succeed ;)