CongducPham/LowCostLoRaGw #174: Receive binary payload

d00m178, 2018-02-24:
Hi.

On my node side payload makes from array of 18 float values.
Then this array sends via LoRa radiochannel.

How to receive this data on gateway side and convert to readable state?

I'm using raw mode:

`sudo ./lora_gateway --raw --bw 125 --cr 45 --sf 7 --freq 915.0 | python post_processing_gw.py | python log_gw.py`

and now I see such output:

```
got first framing byte
--> got LoRa data prefix
raw format from LoRa gateway
Header[dst=1 src=9 seq=6 ptype=0x00]
update ctrl pkt info (^p): 1,9,6,0,72,6,-47
a-8@▒▒`▒X@aD
--- rxlora. dst=0 type=0x00 src=0 seq=0 len=11 SNR=6 RSSIpkt=-39 BW=125 CR=4/5 SF=7
2018-02-24T12:07:41.307702
rcv ctrl pkt info (^p): 0,0,0,0,11,6,-39
splitted in:  [0, 0, 0, 0, 11, 6, -39]
rawFormat(len=11 SNR=6 RSSI=-39)
rcv ctrl radio info (^r): 125,5,7
splitted in:  [125, 5, 7]
(BW=125 CR=5 SF=7)
rcv timestamp (^t): 2018-02-24T12:07:41.306

```

CongducPham, 2018-02-27:
It is much easier, and also size-effective, to send ASCII data. Otherwise, you have to know how data are serialized (endianness) and decode accordingly.

d00m178, 2018-02-27:
I will change code in my nodes in future so they will send data in ASCII, but now I need receive current data in float.
Seems the main issue is decoding my data.
I know that in my payload every 4 byte is one float number, 
so I have 72 bytes of payload total.
I found the place where gateway print received data: