CongducPham/LowCostLoRaGw #176: SyntaxError: EOL ...

kauly, 2018-03-11:
Hi,
I got this error when I use option 2, in the gateway via ssh.

`Running lora_gateway with post-processing... CTRL-C to exit
  File "post_processing_gw.py", line 430
    print 'post status: executing periodic tasks	
SyntaxError: EOL while scanning string literal
END OUTPUT
Press RETURN/ENTER...
`

Option 2 command:

> sudo ./lora_gateway --mode 1 | python post_processing_gw.py

Another question, in gateway_conf.json, the ch is set to -1 I need change this? 
Thanks!

CongducPham, 2018-03-11:
Sorry, it is a bug, it is corrected now, could you take the last version of post_processing_gw.py, or simply edit the file and add a quote, i.e. ' , at the end of line 430.
regards,

CongducPham, 2018-03-11:
For ch, no, don't change it, -1 means use default, which is 865.2MHz. See https://github.com/CongducPham/LowCostLoRaGw#annexa-lora-mode-and-predefined-channels

kauly, 2018-03-11:
Ok, it's working now, thanks. With that command I can see incoming packages right?

CongducPham, 2018-03-11:
yes, but please read here: https://github.com/CongducPham/LowCostLoRaGw/tree/master/gw_full_latest#use-cmdsh-to-interact-with-the-gateway

especially:

---
IMPORTANT NOTICE: Do not launch a new gateway instance with an existing one as there will be conflict on the SPI bus.

You can start manually the gateway for test purposes with option 0.

IMPORTANT NOTICE: To run an operation gateway, it is better to reboot the gateway and let the LoRa gateway program start at boot. Manually lauching the gateway can be usefull for test purposes but we observed that redirections of LoRa gateway output to the post-processing stage can be broken thus leading to a not responding gateway.

You can then use option 5 to see the logs in real time. To test the simple low-level gateway, use option 1. You can ssh at any time and use option 5 to see the latest packets that have been received. 
---

kauly, 2018-03-11:
All is working now, almost crying here kkk. Thank you very much!

closed 2018-03-11 by kauly