CongducPham/LowCostLoRaGw #179: RPi2 and Adafruit RFM9x

MungoBerry, 2018-03-15:
Hi,

I'm trying to create a gateway using a RPi 2 and an Adafruit RFM9x lora module.
I'm using an Arduino UNO for the client.
On Arduino I'll get "Unrecognized transceiver" until I forced to recognize RFM9x module adding on sx1272.cpp at line 263

```
if (version == 0x22) {
        // sx1272
        Serial.println(F("SX1272 detected, starting"));
        _board = SX1272Chip;
    } else
    // for Adafruit RFM9x
    if (version == 0x12) {
        // sx1276
        Serial.println(F("RFM9x detected, starting"));
        _board = SX1276Chip;
    }
    // end  for Adafruit RFM9x
    else
    {
```
otherwise check for sx1276 will fails. It returns version 0.

Now on serial console I get:

> Simple LoRa ping-pong with the gateway
> ATmega328P detected
> RFM9x detected, starting
> SX1276 LF/HF calibration
> ...
> ...
> ...
> Setting Mode: state 1
> Setting Channel: state 1
> Setting Power: state 1
> Setting node addr: state 1
> SX1272 successfully configured
> --> CarrierSense1
> --> CAD duration 300
> ###1
> --> waiting for 1 DIFS (DIFS=3SIFS) = 0
> --> retry
> --> CAD duration 300
> ###1
> --> waiting for 1 DIFS (DIFS=3SIFS) = 0
> --> retry
> --> CAD duration 300
> ###1
> --> waiting for 3 DIFS (DIFS=3SIFS) = 0
> --> retry
> --> CAD duration 300

but I don't know if this output is correct.

On RPi I was unable to run gateway stuff.
I've tried with SD image, then I've tried to recompile it for RPi2.
It hangs at startup on "check version..." inside SPI.transfernb function.
Inside sx1272.h file I 've seen

> #define SX1272_RST  7
> 
> and
> 
> #define SX1272_SS 10

Should not be

> #define SX1272_RST  17
> 
> and
> 
> #define SX1272_SS 8

as reported inside Low-cost-LoRa-GW-step-by-step.pdf ?

I'll try both configuration but still hangs.
Any help ?

Thank you.

CongducPham, 2018-03-15:
Hi, forcing is not going to solve the issue. Probably because the CS pin is not well connected. Check how you connect the CS pin: https://github.com/CongducPham/LowCostLoRaGw#connect-a-radio-module-to-the-raspberry

CongducPham, 2018-03-15:
On the UNO, CS pin should be on pin 10.

MungoBerry, 2018-03-20:
Thank you, solved.
On Arduino I was using bad pin numbers.
On RPi I'll build for version 2, but I've a model B 1.**2**

closed 2018-03-20 by MungoBerry