CongducPham/LowCostLoRaGw #181: LoRa receiver not working on Collar project

ethicstechOSS, 2018-03-19:
Hi. I am evaluating the collar project. I ordered the Fabien Ferrero PCB from pcbway. 
Soldered the LoRa radio & Arduino Pro mini. I downloaded the gps program for collar project, https://github.com/CongducPham/LowCostLoRaGw/tree/master/Arduino/Arduino_LoRa_GPS
Modified the CS pin into 10 for Arduino Pro Mino board and flashed the program. Everything was fine. 

However, I got the error, "Unrecognized receiver..." from the serial monitor.
I just used the same components of what you have been used. But I immediately understand that the PCB you used and the PCB I received is look like some differences, atleast on the appearance. I am comparing the collar document 10th page and attached my PCBs images. PFA. I am not sure this is the reason the LoRa module is not detected or something else. I dont have schematic for this PCB as well. Thanks for all your effort & this great project. Looking forward your help on this issue. I will also report to Fabien Ferrero.

TIA. 
[image]
[image]

ethicstechOSS, 2018-03-19:
Hi. I fixed the issue by adding the  RST pin configuration in sx1272.h library for Arduino Pro Mini configuration,

> #if defined ARDUINO_AVR_PRO || defined ARDUINO_AVR_NANO || defined ARDUINO_AVR_MICRO || defined ARDUINO_AVR_MINI \
> || defined __MK20DX256__ || defined __MKL26Z64__
> #define SX1272_SS 10
> #undef SX1272_RST
> #define SX1272_RST  8

Then I got the following log,

> > Arduino Pro Mini detected
> ATmega328P detected
> SX1276 detected, starting
> SX1276 LF/HF calibration
> ...
> Setting Mode: state 0
> Setting Channel: state 0
> Setting Power: state 0
> Setting node addr: state 0
> SX1272 successfully configured
> Seting HIGH digital GPS power pin: 8
>

CongducPham, 2018-03-21:
Yes, the new version is a bit different, just take as a reference the ANA pin of the RFM95 module which should be connected to the antenna.

ethicstechOSS, 2018-03-22:
Hi. I confirmed that the Radio module 9th pin (antenna PIN ) connected with UCA antenna. However, on this PCB, I also understand that the antenna pin connected with GND as well. Should it be?.

My current output is, the lora module keep on sending its data, but unable to reach/find gateway.

CongducPham, 2018-03-30:
Yes, solder the ground as well, it is better. Did you use PABOOST?

ethicstechOSS, 2018-03-30:
Yes, I used PABOOST.

ethicstechOSS, 2018-03-31:
Hi. Here I shared my log from Arduino Pro Mini and sending simple temperature data on this collar hardware setup for testing. 

> Arduino Pro Mini detected
> ATmega328P detected
> SX1276 detected, starting
> SX1276 LF/HF calibration
> ...
> Get back previous sx1272 config
> Using packet sequence number of 10
> Setting Mode: state 0
> Setting Channel: state 0
> Setting Power: state 0
> Setting node addr: state 0
> SX1272 successfully configured
> Reading 240
> Reading 222
> Reading 213
> Reading 207
> Reading 200
> Reading 195
> Reading 188
> Reading 185
> Reading 176
> Reading 175
> Mean temp is 64.49
> Sending \!#1#TC/21.55567
> Real payload size is 16
> --> CAD duration 301
> ###1

As you can see, the line, "Get back previous sx1272 config" and configured SX1272 instead of SX1276. I am not sure why it is configuring SX1272 instead of SX1276. 

TIA.

ethicstechOSS, 2018-03-31:
Hi. Here I shared my log from Arduino Pro Mini and sending simple temperature data on this collar hardware setup for testing. 

> Arduino Pro Mini detected
> ATmega328P detected
> SX1276 detected, starting
> SX1276 LF/HF calibration
> ...
> Get back previous sx1272 config
> Using packet sequence number of 10
> Setting Mode: state 0
> Setting [...]

As you can see, the line, "Get back previous sx1272 config" and configured SX1272 instead of SX1276. I am not sure why it is configuring SX1272 instead of SX1276. 

TIA.

ethicstechOSS, 2018-03-31:
Hi. I solved the above issue by changing the mode to 4 (on both side) instead of 1 as per the issue, #54. But I am not sure why it is not working with mode 1.

ethicstechOSS, 2018-03-31:
Now testing the GPS project. 
- I did not used mosfet. Directly connected the battery. 
- on the arduino Pro Mini, I can only use Mode 4 with gateway. On Mode 1, its failed to send the data and getting the above issue. 
- I could not able to see the log properly when I connect both, FTDI USB & GPS sensor.
- On the gateway side, I downloaded the latest gw_full_latest, moved into lora_gateway and run lora_gateway with mode 4.

> sudo ./lora_gateway --mode 4
> SX1276 detected, starting.
> SX1276 LF/HF calibration
> ...
> ^$**********Power ON: state 0
> ^$Default sync word: 0x12
> ^$LoRa mode 4
> ^$Setting mode: state 0
> ^$Channel CH_10_868: state 0
> ^$Set LoRa power dBm to 14
> ^$Power: state 0
> ^$Get Preamble Length: state 0
> ^$Preamble Length: 8
> ^$LoRa addr 1: state 0
> ^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
> ^$Low-level gw status ON

I am not sure If I missed anything to run collar project properly. 
Any kind of help appreciated.

TIA.

CongducPham, 2018-04-04:
Without the Mosfet, the GPS module will drain a lot of energy.

As I said, the Pro Mini has only one serial port so connecting the GPS will not allow serial monitor and logs to appear correctly, it may also affect the GPS.

I don't know neither why there is such an issue with mode 1.

Keep me informed. regards,

ethicstechOSS, 2018-04-04:
Thanks for your input. Let me use MOSFET, give a try and get back to you. 
But still I failed to receive any communication (without GPS data) from the node on the gateway.