CongducPham/LowCostLoRaGw #182: Use 868 MHz/915 MHz SX1276 ESP32 Lora

javiisanchez, 2018-03-20:
Hi, we are trying to use 868 MHz MHz SX1276 ESP32 Lora to send message to gateway installed in a raspberrypi with Draguino. The gateway reconize the transeiver and work fine. The problem is when we try to send message from our ESP32, exactly, doesn't recognized the transeiver.
For exemple we are trying this code:

`#include <SPI.h>  
// Include the SX1272
#include "SX1272.h"

#define LCD_SSD1306
#define ESP32_TTGO

#ifdef LCD_SSD1306
#include "SSD1306.h"
SSD1306 display (0x3c, 4, 15);
#endif

// IMPORTANT
///////////////////////////////////////////////////////////////////////////////////////////////////////////
// please uncomment only 1 choice
//
#define ETSI_EUROPE_REGULATION
//#define FCC_US_REGULATION
//#define SENEGAL_REGULATION
/////////////////////////////////////////////////////////////////////////////////////////////////////////// 

// IMPORTANT
///////////////////////////////////////////////////////////////////////////////////////////////////////////
//
// uncomment if your radio is an HopeRF RFM92W, HopeRF RFM95W, Modtronix inAir9B, NiceRF1276
// or you known from the circuit diagram that output use the PABOOST line instead of the RFO line
// Note the Heltec WIFI LoRa needs PABOOST
#define PABOOST
/////////////////////////////////////////////////////////////////////////////////////////////////////////// 

// IMPORTANT
///////////////////////////////////////////////////////////////////////////////////////////////////////////
// please uncomment only 1 choice
#define BAND868
//#define BAND900
//#define BAND433
///////////////////////////////////////////////////////////////////////////////////////////////////////////

#ifdef ETSI_EUROPE_REGULATION
#define MAX_DBM 14
#elif defined SENEGAL_REGULATION
#define MAX_DBM 10
#elif defined FCC_US_REGULATION
#define MAX_DBM 14
#endif

#ifdef BAND868
#ifdef SENEGAL_REGULATION
const uint32_t DEFAULT_CHANNEL=CH_04_868;
#else
const uint32_t DEFAULT_CHANNEL=CH_10_868;
#endif
#elif defined BAND900
const uint32_t DEFAULT_CHANNEL=CH_05_900;
// For HongKong, Japan, Malaysia, Singapore, Thailand, Vietnam: 920.36MHz     
//const uint32_t DEFAULT_CHANNEL=CH_08_900;
#elif defined BAND433
const uint32_t DEFAULT_CHANNEL=CH_00_433;
#endif

///////////////////////////////////////////////////////////////////
// CHANGE HERE THE LORA MODE, NODE ADDRESS 
#define LORAMODE  1
#define node_addr 8
//////////////////////////////////////////////////////////////////
#define PRINTLN                   Serial.println("")
#define PRINT_CSTSTR(fmt,param)   Serial.print(F(param))
#define PRINT_STR(fmt,param)      Serial.print(param)
#define PRINT_VALUE(fmt,param)    Serial.print(param)
#define FLUSHOUTPUT               Serial.flush();

#define DEFAULT_DEST_ADDR 1

uint8_t message[30];

int loraMode=LORAMODE;

void setup()
{
  int e;
  Serial.begin(38400); 

#ifdef LCD_SSD1306
  pinMode(16, OUTPUT);
  digitalWrite(16, LOW);
  delay(50);
  digitalWrite(16, HIGH);

// Initialising the UI will init the display too.
  display.init();
  display.flipScreenVertically();
  display.setFont(ArialMT_Plain_10);
  display.setTextAlignment(TEXT_ALIGN_LEFT);
  display.drawString(5,5,"LoRa Sender");
  display.display();
#endif

  // Print a start message
  PRINT_CSTSTR("%s","Simple LoRa ping-pong with the gateway\n"); 
  
#ifdef LCD_SSD1306
  display.drawString(5, 20, "Simple PingPong");
  display.drawString(5, 35, "LoRa mode 1");   
  display.display();
  delay(1000);
 
#endif

#ifdef ESP32_TTGO
  PRINT_CSTSTR("%s","ESP32 detected\n");
#endif
    
  // Power ON the module
  sx1272.ON();

  // Set transmission mode and print the result
  e = sx1272.setMode(loraMode);
  PRINT_CSTSTR("%s","Setting Mode: state ");
  PRINT_VALUE("%d", e);
  PRINTLN;

  // enable carrier sense
  sx1272._enableCarrierSense=true;
    
  // Select frequency channel
  e = sx1272.setChannel(DEFAULT_CHANNEL);
  PRINT_CSTSTR("%s","Setting Channel: state ");
  PRINT_VALUE("%d", e);
  PRINTLN;
  
  // Select amplifier line; PABOOST or RFO
#ifdef PABOOST
  sx1272._needPABOOST=true;
  // previous way for setting output power
  // powerLevel='x';
#else
  // previous way for setting output power
  // powerLevel='M';  
#endif

  // previous way for setting output power
  // e = sx1272.setPower(powerLevel); 

  e = sx1272.setPowerDBM((uint8_t)MAX_DBM); 
  PRINT_CSTSTR("%s","Setting Power: state ");
  PRINT_VALUE("%d", e);
  PRINTLN;
  
  // Set the node address and print the result
  e = sx1272.setNodeAddress(node_addr);
  PRINT_CSTSTR("%s","Setting node addr: state ");
  PRINT_VALUE("%d", e);
  PRINTLN;
  
  // Print a success message
  PRINT_CSTSTR("%s","SX1272 successfully configured\n");

  delay(500);
}

void loop(void)
{
  uint8_t r_size;
  int e;

  sx1272.CarrierSense();

  sx1272.setPacketType(PKT_TYPE_DATA);
      
  while (1) {
      r_size=sprintf((char*)message, "Ping");
      PRINT_CSTSTR("%s","Sending Ping");  
      PRINTLN;

#ifdef LCD_SSD1306
  display.clear();
  display.drawString(5, 5, "Sending Ping....");
  display.display();
#endif

            
      e = sx1272.sendPacketTimeoutACK(DEFAULT_DEST_ADDR, message, r_size);

      // this is the no-ack version
      // e = sx1272.sendPacketTimeout(DEFAULT_DEST_ADDR, message, r_size);
            
      PRINT_CSTSTR("%s","Packet sent, state ");
      PRINT_VALUE("%d", e);
      PRINTLN;

      if (e==1) {

#ifdef LCD_SSD1306
  display.drawString(5, 20, "Error sending=1 ");
  display.display();
#endif       
      }      
      
      if (e==3) {
          PRINT_CSTSTR("%s","No Pong from gw!");
  
#ifdef LCD_SSD1306
  display.drawString(5, 35, "No Pong from gw!");
  display.display();
#endif          
      }

      if (e==1 || e==3) {
#ifdef LCD_SSD1306
  display.drawString(5, 35, "SNR at gw=N/A");
  display.display();
#endif           
      }
        
      if (e==0) {
          char msg[20];
          sprintf(msg,"SNR at gw=%d   ", sx1272._rcv_snr_in_ack);
          PRINT_CSTSTR("%s","Pong received from gateway!");
          PRINTLN;
          PRINT_STR("%s", msg);   
#ifdef LCD_SSD1306
  display.drawString(5, 35, "Get Pong from gw");
  display.drawString(5, 35, msg);
  display.display();
#endif   
      }

      PRINTLN;
      
      delay(10000);   
  }          
}`

and in the SX1272.h we have added;

`//it is not mandatory to wire this pin
//we take pin 4 as it is available on many boards
#ifdef ESP32_TTGO
#define SX1272_RST  14
#else
#define SX1272_RST  4
#endif

#if defined ARDUINO_AVR_FEATHER32U4 || defined ARDUINO_SAMD_FEATHER_M0
// on the Adafruit Feather, the RFM95W is embeded and CS pin is normally on pin 8
#define SX1272_SS 8
#elif defined ARDUINO_ESP8266_ESP01
#define SX1272_SS 15
#elif defined ESP32_TTGO
#define SX1272_SS 18
#else
// starting from November 3rd, 2017, the CS pin is always pin number 10 on Arduino boards
// if you use the Libelium Multiprotocol shield to connect a Libelium LoRa then change the CS pin to pin 2
#define SX1272_SS 10
#endif`

Can you help us? 
Our transeiver use the followinf SPI pins:
SCK--5
RST--14
CS--18
MOSI--27
MISO--19 
IRQ--26
 
Thanks in advance.

CongducPham, 2018-03-20:
The problem is that your define ESP32_TTGO is probably locally defined, so when SX1272.cpp is compiled, the define statement is not declared, so not assumed to be not defined. It is better to use the setCSPin() function as in this example: https://github.com/CongducPham/LowCostLoRaGw/blob/75d3bbb597339f1ccdd6ecb0422236a4d3999b7d/Arduino/Arduino_LoRa_Ping_Pong_LCD/Arduino_LoRa_Ping_Pong_LCD.ino#L227

Get the latest Arduino/sketch/libraries/SX1272 lib

javiisanchez, 2018-03-20:
Yes, I also have tried to add this line, but it follow unrecongnizing the transeiver. (without modifying the SX1272.h
```
#ifdef ESP32_TTGO
  PRINT_CSTSTR("%s","ESP32 detected\n");
  sx1272.setCSPin(18);
#endif
    
  // Power ON the module
  sx1272.ON();
```
Also, we have tried to define ESP32_TTGO gloablly. but it doesn't work.

CongducPham, 2018-03-21:
So, check again the connection. I don't see what can be the other alternatives. The RST pin is not used.

regards,

javiisanchez, 2018-03-21:
We can check the connections but the connections it was made at the PCB of the ESP32 who we can see in the following picture:
[image]

CongducPham, 2018-03-21:
I don't see the MISO pin, maybe you inverted MOSI and MISO?

javiisanchez, 2018-03-21:
Yes in the picture doesn't specify the MISO pin, surely its a error.
The MISO is pin 19 and the MOSI the 27.
what pins do you define or configure by default?

CongducPham, 2018-03-21:
These pins are defined by the board, so we are not re-defined them, only CS can be "user" defined. I don't have any answer, maybe a hardware issue, can you try with another board?

javiisanchez, 2018-03-22:
Yes I can check with other device. Now, I'm verify the hardware and this is the finally schematic:
[image]

javiisanchez, 2018-03-22:
Hi,  after trying different solutions we have aaccomplished. We have modified the ON() method in SX1272.cpp, we have changed the SPI.begin() by SPI.begin(SX1276_SCK, SX1276_MISO, SX1276_MOSI, SX1276_SS);
SCK--5
RST--14
CS--18
MOSI--27
MISO--19
```
//Configure the MISO, MOSI, CS, SPCR.
		#ifdef ESP32_TTGO
			SPI.begin(SX1276_SCK, SX1276_MISO, SX1276_MOSI, SX1276_SS);
		#else
	    SPI.begin();
		#endif
```

if you want we can do a commit to add this board to the framework. (It we also have added the LCD of the board  in the ping_pong_LCD)

Thanks.

CongducPham, 2018-03-22:
Thank you. I guess it is because the Arduino IDE does not have the TTGO board variant definition so the pins_arduino.h file is not defined for this board. I found this githhub that seems to be for the TTGO board: https://github.com/CaptIgmu/Arduino/tree/master/esp32/TTGO_LoRa32_OLED

Maybe this can be a solution to support the board. As you can see the board.txt file defines the board as ARDUINO_TTGO_LoRa_32.

romax1407, 2018-06-21:
[image]