CongducPham/LowCostLoRaGw #183: Bunch of header files not found

rizkyarlin, 2018-03-27:
Thanks for the very comprehensive tutorial. Unfortunately, I got a problem with the libraries.

I just added the gw_full_latest library to my Arduino IDE. In the first compiling netinet/in.h not found, so I added it manually to my libraries. The second compile showed that sys/ipc.h is not found, again I tried to added manually. Then it showed sys/sem.h not found. Now I'm sure there will be dozens of it.

I am running Arduino IDE in my Windows 7. My friend is using raspberry pi and everything is okay. Can you point me how to solve this problems?

CongducPham, 2018-03-27:
The gateway program is only for Raspberry. If you want to use another platform, you may read section "7. Tools for testing & build a LoRa gateway for on-the-go tests" at http://cpham.perso.univ-pau.fr/LORA/RPIgateway.html but not all features can be realized.

regards,

rizkyarlin, 2018-03-27:
Thanks for the quick response :D 
Do you have a simple example of sending a simple data from one node to another node?