CongducPham/LowCostLoRaGw #185: Unrecognized transceiver

XRobquin, 2018-04-08:
Hi !
I am a beginner in arduino, and I have to use SX1272 on InAIR9B with an Arduino Uno.
Only, once the connections between the Uno and the LoRa module are done, I get the following message : 
"Unrecognized transceiver"
The connections I have made are the following : 

UNO - InAIR9B
gnd - 0V
3V3 - 3V3
MISO - MISO
MOSI - MOSI
CLK - CLK
Reset - Rst
Pin 10 - Chip Select
Pin 6 - D2
Pin 5 - D1
Pin 2 - D0

What can I do to get rid of that error ?

Thanks in advance for your help
XRobquin

CongducPham, 2018-04-14:
Try to not connect D2, D1 and D0, our lib does not use them. Also, do not connect the Arduino RESET pin. And also, what are the pins you are using on the Uno for MISO, MOSI and CLK?

XRobquin, 2018-05-04:
Thank you very much, it works !

CongducPham, 2018-05-04:
Great to hear! All the best for your project.