CongducPham/LowCostLoRaGw #186: Question, payload length/type

kauly, 2018-04-13:
Can I send an array to the gateway? Exist a payload max size? If so the library broken the data into various  packages?

Thanks.

CongducPham, 2018-04-13:
You have to do all the data type management yourself, including serialization/deserialization, big/little endianess if necessary.

regards,

kauly, 2018-04-14:
Thanks, I will pass the arr to string.

closed 2018-04-14 by kauly

kauly, 2018-04-16:
Solution:
```c
float dados_arr[24];

String arr_to_str(float * arr_p){
    String str = String("\\!");
    for (int i = 0;i < 24;i++){
        String valor = String(arr_p[i]);
        //delimitador ">"
        if(i < 23)
          valor.concat(">");
        str.concat(valor);
    }
    return str;
}
void envia_dado(String dados_p){  
  uint8_t r_size = dados_p.length()+1;
  uint8_t message[r_size];
  dados_p.toCharArray((char *)message, r_size);
  int e;
  sx1272.CarrierSense();
  sx1272.setPacketType(PKT_TYPE_DATA);
  e = sx1272.sendPacketTimeout(DEFAULT_DEST_ADDR, message, r_size);
  Serial.print("Dado enviado, estado: ");
  Serial.println(e);
 // delay(1000*1000);
  delay(1000);
}

envia_dado(arr_to_str(dados_arr));
//or send ordinary data
envia_dado("Hello GW")
```
Maybe this help someone.