CongducPham/LowCostLoRaGw #187: Clouds Scripts

kauly, 2018-04-19:
Hi, one more question 
It's possible use paho.mqtt instead of cmd commands? Like this;

```
import paho.mqtt.publish as publish
import sys

servidor = "xxx"

def main(ldata, pdata, rdata, tdata, gwid):
    arr = map(int,pdata.split(','))				
    src=arr[2]
    dados_obj = {
        'node': src,
        'dados': ldata,
        'timestamp': tdata,
        'gw_id': gwid
    }
    publish.single("IFSC_SJ/Pluviometro/Dados", dados_obj, hostname=servidor)
		
if __name__ == "__main__":
	main(sys.argv[1], sys.argv[2], sys.argv[3], sys.argv[4], sys.argv[5])
```

CongducPham, 2018-04-19:
Yes, it is possible to use the paho python mqtt package if you prefer.

kauly, 2018-04-19:
Thanks again! The code has a typo, dados_obj must be a string, str(dados_obj).

closed 2018-04-19 by kauly