CongducPham/LowCostLoRaGw #193: send rssi value

surender128, 2018-05-14:
good morning..
  how to send rssi value in arduino program along with data

CongducPham, 2018-05-14:
rssi is available when you receive a packet. Why would you send along with data? Do you receive before sending data?

After a packet reception, call sx1272.getRSSIpacket() then the value is stored in sx1272._RSSIpacket

regards,

surender128, 2018-05-14:
thank you for your reply..
i want to print the rssi value on oled which is connected to arduino.
thank you..
now i got..

surender128, 2018-05-25:
[image]
 

hi ,
    in above you told the procedure to read the rssi value.but when i use that in program the rssi value is not same in transmitter(lora) and receiver(raspberrypi). please help me to understand

CongducPham, 2018-05-27:
RSSI can be of course different at transmitter and receiver because they are not measured at the same time. In a any way, even if measured at the same time, there is always fluctuation so do not expect to have same value. At the device, before transmission, the RSSI can be additionaly measured (in addition to the Channel Activity Detection mechanism) to detect whether there is activity or not. If the RSSI is below a given value then we assume there is no activity. Then at the gateway, we measure the RSSIpkt of the received packet, so of course we expect that this RSSIpkt is higher.