CongducPham/LowCostLoRaGw #194: MongoDB error in date

zhapwey, 2018-05-15:
hello:

I've got everything to work perfectly. Great work on the part of the developers.
I only find one thing that can not be fixed. The mongodb database saves the data correctly but with the UTP time not with the local time, on the other hand in nodered it makes it perfect.
It can be fixed ???

CongducPham, 2018-05-16:
You can do that by replacing in line 195 of CloudMongoDB.py (https://github.com/CongducPham/LowCostLoRaGw/blob/66715ac1442a9263fadd6ef0c15c342f3710d6c1/gw_full_latest/CloudMongoDB.py#L195)

"time":now, by "time":tdata,

man4567890, 2019-07-28:
Hi
The change does not seem to work, I get data from the 1970's then, see screenshot below
[image]
pi@raspberry:/var/log/mongodb $ sudo cat /home/pi/lora_gateway/CloudMongoDB.py | grep tdata
def main(ldata, pdata, rdata, tdata, gwid):
		"time":tdata,
pi@raspberry:/var/log/mongodb $

man4567890, 2019-08-08:
Hi,
Is this issue logging still active? Could you see if you can look at the MongoDB problem please, my time is still being reported as Zulu time and not my local timezone
Regards