CongducPham/LowCostLoRaGw #198: Max messages whit gps

Kronez, 2018-05-20:
Hi sir

First thank you for great project you share.
I install gw on rpi and make pingpong test whit Arduino. All work perfect. Next i build that gps transmitter and i got it work as well. After i use ot one day i notes that gw only receive 100 message from arduino and after that nothing. if i restart arduino and beacon counter start from 0 it works again 100 messages. I check that arduino send radio pulse normal after 100 but gw wont receive. any idea how to fix it?

CongducPham, 2018-05-22:
Let me try to check this issue. I'll come back to you.

CongducPham, 2018-06-26:
We have not been able to reproduce this behavior. It is working fine for us.