CongducPham/LowCostLoRaGw #201: Radio Module not sending ping

UrjitDave, 2018-05-30:
Hi @CongducPham Sir 
I'm using EthicsTech LoRa Development Kit (http://www.ethicstech.in/products/iot/lora-iot) and trying to run Arduino_Lora_Ping_Pong program from your sdk. We have a LoRa Gateway configured with Radio Module as per your tutorial. But it is still not receiving anything.
[image]

I am uploading this sketch : 
[https://github.com/CongducPham/LowCostLoRaGw/tree/master/Arduino/Arduino_LoRa_Ping_Pong
](url)

The Serial Monitor of Arduino IDE displays : 
[image]

Looking forward to your response

CongducPham, 2018-05-30:
Check the CS pin on this board, we use pin 10 as default, maybe it is different for this board.

UrjitDave, 2018-05-30:
Thanks for the prompt reply. The board that we are using is a **combination of Arduino and LoRa Radio Module** and runs the Arduino sketches itself. Thus I don't need to connect any boards for sending packets. Here's how my board look :
[image]

The board, when attached to Arduino IDE, displays as "Adafruit Feather M0" and runs other sketches smoothly. But the Ping Pong sketch is uploaded it shows  ... ... in Serial Monitor.

What could be the issue then ?

CongducPham, 2018-05-30:
As I said, the CS pin used by this board may be not pin 10, so check the board schematic. If it is like the Adafruit M0, then maybe pin 8. Either change in SX1272.h or do as shown here: https://github.com/CongducPham/LowCostLoRaGw/blob/43c49579248f5599b5e55425d939df3d9a2c8d90/Arduino/Arduino_LoRa_Ping_Pong_LCD/Arduino_LoRa_Ping_Pong_LCD.ino#L227

CongducPham, 2018-05-30:
If you look at the vendor github https://github.com/ethicstechOSS/LoRa-Development-Kit, the CS pin is pin 5, so use sx1272.setCSPin(5);

UrjitDave, 2018-05-31:
Thanks, Sir for the prompt reply. I tried  sx1272.setCSPin(5), and the serial monitor displays:

[image]
 
But the Gateway still displays : 
[image]

I cannot see the packets received on the gateway. However, when I press 'l' on the gateway interface, it displays the following two entries. I don't know how I got those 
[image]

What could be the issue ??