CongducPham/LowCostLoRaGw #206: FIFO behavior on Hoperf RFM96 LoRa transceiver

ElColorino, 2018-06-10:
Has anyone noticed that, contrary to the Hoperf document for this device, setting LORA_SLEEP_MODE does not clear the FIFO. It seems to set it to random values when I do this. I have tried various mode transitions, since FSK mode explains how that FIFO gets cleared in more detail than the LoRa FIFO. The only way I have found to clear the FIFO to zero, besides writing zeros to it, is to assert the RESET signal to the chip. Has anyone else seen this behavior?

CongducPham, 2018-06-14:
Hi, sorry for my late answer, actually the FIFO is not cleared, when you receive new message, you have to set the pointer to where is the beginning of your FIFO. Then since you know how many bytes are received, you are reading only the correct number of bytes.

ElColorino, 2018-06-14:
Professor
Thanks for your reply. The actual product engineer from Hoperf also replied
and confirmed what you just told me, and that their document is incorrect.
I suspect it is a translation-to-English problem. They write about the FIFO
being "cleared" but what they actually mean is that, after certain mode
transitions, the contents are indeterminate. I am writing test code for the
Loranga board and testing FIFO behavior is where this issue came up.

Best regards,

Richard Murphy

ElColorino, 2018-06-17:
Professor
Another question about the SX1272 code in your lora_gateway github release. Can you actually poll the IRQ register using SPI? When I do that when in LORA_TX_MODE, I get bizarre results. If I simply wait 500ms and check, everything looks normal.

Richard

CongducPham, 2018-06-20:
If you look at the code, then yes it is polling the register to detect the TX_DONE interrrupt, but as there are some other instructions, there maybe some delay between 2 read.