CongducPham/LowCostLoRaGw #210: No ACK from Gateway

aniket1002, 2018-06-19:
hello @CongducPham sir,
I have an issue running the ping_pong code. 
When i run it using no-ACK version of the code I receive the following on my serial monitor:
 
[image]

And when I run it with ACK I get the following on the serial monitor:
[image]
Although in both the cases my gateway displays this:
[image]

what can be the issue here?
how can i receive packets with ACK in ping_pong code?
Looking forward to your response.
Regards.

jtroeth1, 2018-06-26:
Without ACK, there is no way of knowing if the message is received, hence the Serial monitor says pong received. With ACK, the pro mini is waiting for a reply to confirm that the message was received. In both cases the message is not received by the gateway.

CongducPham, 2018-06-26:
There are several issues I see from your screenshot:

1/ there is state=2 when setting power at device, it is not normal.
2/ cr=4 is not a valid option at gw, the minimum is 5 to have the 4/5 mode.

Please be sure to have the same setting at both ends.