CongducPham/LowCostLoRaGw #211: Lora_gateway program running CPU usage reached 100%

ylzmdmoon, 2018-06-22:
Lora_gateway program running CPU usage reached 100%;
./lora_gateway Starting with a separate gateway or starting with post_processing_gw.py, the CPU usage is almost 100%. I am running on raspiberry 3B+. Is this normal?

CongducPham, 2018-06-22:
Yes, as the RPI constantly listen for radio packet at both low-level and high-level (post-processing). However, it doesn't mean that you cant' launch other program. We are trying to optimize the way the low-level and the high-level gateway interact each other.

ylzmdmoon, 2018-06-23:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
thank you for your reply!I use the gps hat as a gateway to lora, and collect gps information to the gateway. The gps hat is installed on the Raspberry Pi. In the case of high CPU usage, the heat is severe. Will this affect the performance of the gateway?

    
        a#ntes-pcmail-signature-default:hover {
            text-decoration: underline;
            color: #199cff;
            cursor: pointer;
        }
        a#ntes-pcmail-signature-default:active {
            text-decoration: underline;
            color: #246fce;
            cursor: pointer;
        }
    

                
                    发自网易邮箱大师

CongducPham, 2018-06-25:
We have several gateways running for a long time. Havent' noticed any overheating.
regards,

rertini, 2018-08-26:
Putting a small delay i.e. delay(1) in the first while loop of function availabledata(...) lowers the CPU usage to 4~5%:

...
        while( (bitRead(value, 4) == 0) && (millis() < exitTime) )
        {
            value = readRegister(REG_IRQ_FLAGS);
// put delay here
delay(1);     
            // Condition to avoid an overflow (DO NOT REMOVE)
 ...

CongducPham, 2018-08-26:
Thanks for the suggestion!

regards,

CongducPham, 2018-08-26:
We may provide a compiler option to use DIO interrupt to lower CPU work.

rertini, 2018-08-26:
Yes, that would be much better.