CongducPham/LowCostLoRaGw #212: mqtt with username and password

htcheroportugal, 2018-06-24:
hello, my broker mqtt had username and password, how can use my private broker with this gateway?

CongducPham, 2018-06-24:
CloudMQTT.py uses mosquitto_pub, just add your credentials using options

htcheroportugal, 2018-06-24:
@CongducPham sorry ask you again, but can you show me how can to do that? i dont undersand what you say.

htcheroportugal, 2018-06-25:
@congducPham if nomenclatures=='':
	cmd = 'mosquitto_pub -u username -P password -h '+key_MQTT.MQTT_server+' -t '+data[0]+'/'+data[1]+'/'+src+' -m \"'+data[i+2]+'\"'
else:
	cmd = 'mosquitto_pub -u username -P password -h '+key_MQTT.MQTT_server+' -t '+data[0]+'/'+data[1]+'/'+src+'/'+nomenclatures[i]+' -m \"'+data[i+2]+'\"'

That is correct?

CongducPham, 2018-06-25:
Looks good.

htcheroportugal, 2018-06-25:
@CongducPham  i tried with:

[image]

But not sucess.my broker is 192.168.1.855

CongducPham, 2018-06-26:
I believe you replace username and password with your actual user name and password right?

htcheroportugal, 2018-07-01:
no, sorry i forgot that...  i replaced it and now it works.. thanks again for your support.

closed 2018-07-01 by htcheroportugal