CongducPham/LowCostLoRaGw #213: Only Receiving every second ping at gateway end

jtroeth1, 2018-06-26:
Hi there,

I am running the ping pong code and I am having an issue with getting the gateway to receive the ping when the delay is anymore than 7 seconds (at pro mini node end). Below 7 seconds and I have no issues (receive 100% of messages). Its as if the gateway is in a sleep state and doesn't wake when a message comes in. (possibly below 7 seconds means it is not entering a sleep mode??).

I might be able to get around this by sending two messages, or by sending the messages frequently, however I still think its an issue I would like to resolve.

Thanks

CongducPham, 2018-06-26:
Well, we haven't encouter this issue so I don't know what could be the reason. Did you check that the SNR of received messages is good, typically greater than 0?

CongducPham, 2018-06-30:
I don't think so, it may be another issue.

bretskee, 2019-11-05:
@jtroeth1 did you solve this issue? I am having the same problem. It is not an SNR issue and happens consistently every second time at 10 seconds. As you suggest there are repetition windows where it works ever time and those every
 second time.

jtroeth1, 2019-11-06:
@bretskee I did not. I ended up writing my own software as the code was a little unreliable for my application.

CongducPham, 2019-11-06:
Hi, I still don't have this issue, regardless of the time between 2 ping. It is working fine on my gateway. Did you check on the gateway that "raw" option is false to enable ping request?

bretskee, 2019-11-11:
Raw is set to false. If it was not, i would never get a response. I get a response every second time.

CongducPham, 2019-11-11:
Just to be sure to understant. What do you mean by "every second time at 10 seconds"?

regards,