CongducPham/LowCostLoRaGw #215: set max 20dBm in gateway

surender128, 2018-07-03:
Hi,
    I read the SX1272.cpp and SX1721.h files.
In that i dont know how to set the 20dBm bye calling 'X'. please help me for to change

CongducPham, 2018-07-03:
Simply with either setPowerDBM(20) or setPower('X')

surender128, 2018-07-03:
thank you for your reply...
where i can change this function?

surender128, 2018-07-04:
[image]

i uncommented #define PABOOST
after that i changed setPower('X')
but in the log of the gateway dBm value is not changed

[image]

thank you

surender128, 2018-07-04:
[image]

i uncommented #define PABOOST
after that i changed       loraPower='X'; (in 1st screenshot)
but in my gateway  log the value about dbm is not changed

[image]

thank you.

CongducPham, 2018-07-04:
For the gateway side, read this: https://github.com/CongducPham/LowCostLoRaGw/tree/master/gw_full_latest#compiling-the-low-level-gateway-program

So you should have something like:

CFLAGS=-DPABOOST -DMAX_DBM=20