CongducPham/LowCostLoRaGw #217: Unexpectedly the code doesn't run, and after throws unrecognized transceiver

sagilar, 2018-07-06:
Hello,

I am working with the library with a Raspberry PI 3 and a Dragino LoRa HAT  915 MHz. I followed the steps given by the Dragino page in which indicates it is needed to sold pin 22 to pin 24 in the HAT.
It was working ok, when I compiled the code and I ran the code (sudo ./lora_gateway), the console threw "SX 1276 detected, starting ...". But now (I don't know why), I ran the sudo ./lora_gateway command, and the console throws nothing. It just stays in blank, and after much time (sometimes, sometimes doesn't throw anything) it throws "Unrecognized Transceiver ...".
The weird thing is that when I run the sudo ./lora_gateway_pi2 (An executable file in previous version), the console throws the correct output ("SX 1276 detected, starting ..."), but the ./lora_gateway never works.

I don't know what to do, please help me.
Thank you in advance for your help.

[image]

CongducPham, 2018-07-06:
You have to compile for RPI3:

> make lora_gateway_pi2

Otherwise, run the basic_config_gw.sh scripts that does that for you.

sagilar, 2018-07-06:
Thanks, it works.

I have another question.

I am trying to connect some STM LoRa DISCO L072CZ LRWAN1 devices to the Rpi+Dragino Gateway, but I haven't been able to communicate them with the gateway.
I have made some tests to connect between STM devices and it worked. It also worked when I used the single_chan_pkt_fwd library (It works but I got garbage in the "data" parameter).
Is it any protocol to link these different devices? or what change can I do to enable the communication.

P.S: I am programming the STM devices using MBED.
Thanks in advance.

sagilar, 2018-07-06:
I forgot to mention that I am using the 915.0000 MHz in the STM devices, so I change the Gateway 900MHz band to channel 12, as referenced in your tutorial.