CongducPham/LowCostLoRaGw #218: RadioHead send results in ** The CRC is incorrect **

pointhi, 2018-07-07:
I use a simple sender with the default Modem configuration: ```Bw = 125 kHz, Cr = 4/5, Sf = 128chips/symbol```. I do not use the ```RHDatagram``` as in your example, but set the ids directly. This shouldn't be the problem though.

I enabled debug mode in the SX1272 library to find the problem:
```c
#define SX1272_debug_mode 1
```

When I run the gateway, I now get the following output. Changing Coding Rate does not solve the issue, and different SF and BW result in no Receive at all.
```
$ sudo ./lora_gateway_pi2 --freq=433.3 --raw --sf=7 --bw=125 --cr=5
Start Setup.
SX1276 detected, starting.
SX1276 LF/HF calibration
...
^$**********Power ON: state 0
^$Default sync word: 0x12
^$LoRa CR 5: state 0
^$LoRa SF 7: state 0
^$LoRa BW 125: state 0
^$Frequency 433.300000: state 0
^$Set LoRa power dBm to 14
^$Power: state 0
^$Get Preamble Length: state 0
^$Preamble Length: 8
^$LoRa addr 1: state 0
^$Raw format, not assuming any header in reception
^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
Setup Finished.
^$Low-level gw status ON

Starting 'availableData'
## Valid Header received in LoRa mode ##
## Checking destination ##
## Packet received is for me ##

Starting 'getPacket'
## Packet received in LoRa mode ##
** The CRC is incorrect **
getPacket() gives state=3
^$Receive error 4

Starting 'availableData'
## Valid Header received in LoRa mode ##
## Checking destination ##
## Packet received is for me ##

Starting 'getPacket'
## Packet received in LoRa mode ##
** The CRC is incorrect **
getPacket() gives state=3
^$Receive error 4
```

Board is at the moment the [Draguino Lora/GPS_HAT](https://wiki.dragino.com/index.php?title=Lora/GPS_HAT) with the recommended solder bridge. After looking into the SourceCode I found out it could be solved with a little configuration change of ```SX1272.h``` as well, but nevermind.

The Link is working correctly. I wrote my own gateway software some time ago ([pointhi/lora_single_chan_gateway](https://github.com/pointhi/lora_single_chan_gateway)) and it is able to receive all packages as expected. It is not visible in this log, but all CRC Checks succeded, so the hardware should be not at fault:

```
./lora_single_chan_gateway/gateway.py 
INFO:root:SX1276 detected, starting...
INFO:root:Listening at SF7 on 433.3 MHz
INFO:root:Received at 2018-06-19T12:21:16.570401+00:00: ff10cb001020cdcccc41220041ce47802b87564090a69bc43c = b'\x10 \xcd\xcc\xccA"\x00A\xceG\x80+\x87V@\x90\xa6\x9b\xc4<'
INFO:root:Received at 2018-06-19T12:21:21.698677+00:00: ff10cc001020cdcccc41220044ce4780b6f3554090a69b443d = b'\x10 \xcd\xcc\xccA"\x00D\xceG\x80\xb6\xf3U@\x90\xa6\x9bD='
INFO:root:Received at 2018-06-19T12:21:26.825928+00:00: ff10cd001020cdcccc41220043ce4780df4f554090a69b443d = b'\x10 \xcd\xcc\xccA"\x00C\xceG\x80\xdfOU@\x90\xa6\x9bD='
INFO:root:Received at 2018-06-19T12:21:31.954215+00:00: ff10ce0010200000cc41220042ce4780df4f554090a69b443d = b'\x10 \x00\x00\xccA"\x00B\xceG\x80\xdfOU@\x90\xa6\x9bD='
```

CongducPham, 2018-07-09:
Strange, let me check this and I'll get back to you.

pointhi, 2018-07-14:
@CongducPham updating the gateway solved the issue :+1: 

Probably it would be a good idea to update the raspberry-pi image as well

closed 2018-07-14 by pointhi