CongducPham/LowCostLoRaGw #226: Arduino_LoRa_Simple_temp	issue the end-device stop sending temperature

xfocus99, 2018-09-07:
hello,
i am using arduino mini pro 3V3 8Ghz.

i have those issues.

- after some time the GW is not receiving any data from the end-device.

- some time the GW receives data after every 20min even the cycle is set to 10min.

CongducPham, 2018-09-07:
Check that PA_BOOST is set accordingly (SNR should be positive). If 20min maybe you lost a packet which is quite normal.

xfocus99, 2018-09-07:
i have the modtronix Wireless SX1276 LoRa Module, 868MHz and 915MHz.
the PA_BOOST is disabled. 
i did my test indoor i did put the GW(Raspberry pi pi 3 b+modtronix inair9) in a rom and the end-device (modtronix inair9 + arduino mini pro+battery) in the next troom.

when i did put the end-devis outside the house(+50m) the GW (inside the house)didn't receive any data from the end-device(it should work for at least +500m). when i return the end-device to the next room i have to reboot end-device( power of then power on) so it can send data, without this trick it will never send data.

xfocus99, 2018-09-10:
hi again thank you for your help, i did check the parametre and it 'unsigned int idlePeriodInMin = 10;'
i have an inAir9, now the node is transmiting periodically every 20min even if it fixed for 10min.
i have the last update for the GW and the node.
i am using DS18B20 water temperature.
i will upload some images ASAP.
but my question is why they said it low power long range IoT Gw and Node but it stop working for a range more than +50m indoor.

i didn't test it outdoor, is this need to be with a clear LoS to have long range?

CongducPham, 2018-09-10:
Normally, the range is about 2 to 4 kms. We tested that in several real-world deployment. So there is an issue with your setting but I don't know what.

xfocus99, 2018-09-11:
i think the problem is in the node settings. because the GW settings are unchanged after new installation.i have an antenna of 915MHz for both the Gw and the node.
here is my node sketch:

`/*
 *  temperature sensor on analog A0 to test the LoRa gateway
 *
 *  Copyright (C) 2016-2018 Congduc Pham, University of Pau, France
 *
 *  This program is free software: you can redistribute it and/or modify
 *  it under the terms of the GNU General Public License as published by
 *  the Free Software Foundation, either version 3 of the License, or
 *  (at your option) any later version.

 *  This program is distributed in the hope that it will be useful,
 *  but WITHOUT ANY WARRANTY; without even the implied warranty of
 *  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *  GNU General Public License for more details.
 *
 *  You should have received a copy of the GNU General Public License
 *  along with the program.  If not, see <http://www.gnu.org/licenses/>.
 *
 *****************************************************************************
 * last update: June 29th, 2018 by C. Pham
 */

// IMPORTANT
///////////////////////////////////////////////////////////////////////////////////////////////////////////
// add here some specific board define statements if you want to implement user-defined specific settings
// A/ LoRa radio node from IoTMCU: https://www.tindie.com/products/IOTMCU/lora-radio-node-v10/
//#define IOTMCU_LORA_RADIO_NODE
///////////////////////////////////////////////////////////////////////////////////////////////////////////
#include <OneWire.h>
#include <DallasTemperature.h>

#include <SPI.h> 
// Include the SX1272
#include "SX1272.h"
//uncomment if you want to disable WiFi on ESP8266 boards
//#include <ESP8266WiFi.h>
#define ONE_WIRE_BUS 8
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);

// IMPORTANT
///////////////////////////////////////////////////////////////////////////////////////////////////////////
// please uncomment only 1 choice
//
#define ETSI_EUROPE_REGULATION
//#define FCC_US_REGULATION
//#define SENEGAL_REGULATION
/////////////////////////////////////////////////////////////////////////////////////////////////////////// 

// IMPORTANT
///////////////////////////////////////////////////////////////////////////////////////////////////////////
// please uncomment only 1 choice
#define BAND868
//#define BAND900
//#define BAND433
///////////////////////////////////////////////////////////////////////////////////////////////////////////

#ifdef ETSI_EUROPE_REGULATION
#define MAX_DBM 14
// previous way for setting output power
// char powerLevel='M';
#elif defined SENEGAL_REGULATION
#define MAX_DBM 10
// previous way for setting output power
// 'H' is actually 6dBm, so better to use the new way to set output power
// char powerLevel='H';
#elif defined FCC_US_REGULATION
[... 610 lines cut ...]
          //timer is triggered
          ESP.deepSleep(LOW_POWER_PERIOD*1000*1000);
#else
          // use the delay function
          delay(LOW_POWER_PERIOD*1000);
#endif                        
          PRINT_CSTSTR("%s",".");
          FLUSHOUTPUT
          delay(10);                        
      }
      
      delay(50);
#endif      
      
#else
      PRINT_VALUE("%ld", nextTransmissionTime);
      PRINTLN;
      PRINT_CSTSTR("%s","Will send next value at\n");
      // use a random part also to avoid collision
      nextTransmissionTime=millis()+(unsigned long)idlePeriodInMin*60*1000+(unsigned long)random(15,60)*1000;
      PRINT_VALUE("%ld", nextTransmissionTime);
      PRINTLN;
  }
#endif
}`

xfocus99, 2018-09-14:
here is the pictures of my project.

https://imgur.com/a/ijsPNQz

thank you for your help.

xfocus99, 2018-09-16:
here is my stats for 5m range: 

2018-09-16T15:06:09.947098> --- rxlora. dst=1 type=0x12 src=6 seq=49 len=17 SNR=-13 RSSIpkt=-114 BW=125 CR=4/5 SF=12
2018-09-16T15:06:09.947794> 2018-09-16T15:06:09.942823
2018-09-16T15:06:09.948305> rcv ctrl pkt info (^p): 1,18,6,49,17,-13,-114
2018-09-16T15:06:09.948782> splitted in:  [1, 18, 6, 49, 17, -13, -114]
2018-09-16T15:06:09.949229> (dst=1 type=0x12(DATA WAPPKEY) src=6 seq=49 len=17 SNR=-13 RSSI=-114)

CongducPham, 2018-09-16:
The SNR is too low, which usuallt means at short range that PA_BOOST setting is now correct. I see in your code that #define PABOOST is uncommented. If you have the inAir9, then you don't need PA_BOOST so comment this line.

xfocus99, 2018-09-16:
do you think that with this setting i can get long range 2km LoS?

closed 2018-09-20 by xfocus99