CongducPham/LowCostLoRaGw #230: Problem: Node device did not receive message from gateway

TienPham77777, 2018-10-05:
Dear @CongducPham ,

The first i wanna tell that your project so awesome is! I've just begun to learn how to build lora network. Now i try to test downlink feature by using LoRa_InteractiveDevice source to send "hello" message to gateway (following this link [https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-downlink.md](https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-downlink.md)).
After had received about 5s, the gateway sends back "reply from gw" but node device receives nothing.

The address of the device is 6 and send "hello" to the gateway:

> SX1276 detected, starting
SX1276 LF/HF calibration
...
^$**********Power ON: state 0
^$Default sync word: 0x12
^$LoRa mode 1
^$Setting mode: state 0
^$Channel CH_0_433: state 0
^$Use PA_BOOST amplifier line
^$Set LoRa power dBm to 14
^$Power: state 0
^$Get Preamble Length: state 0
^$Preamble Length: 8
^$LoRa addr 6: state 0
^$CRC 1: state 0
^$CRC 1: state 0
^$SX1272/76 configured as device. Waiting serial input for serial-RF bridge
Rcv serial: hello
Sending. Length is 7
hello
Packet number 0
Payload size is 7
ToA is w/4B header 1287
--> CarrierSense2: do CAD for DIFS=9CAD
--> DIFS duration 548
OK1
LoRa Sent in 1406
LoRa Sent w/CAD in 1956
Packet sent, state 0

Gateway's output:

> --- rxlora. dst=1 type=0x10 src=6 seq=3 len=7 SNR=7 RSSIpkt=-15 BW=125 CR=4/5 SF=12
2018-10-05T11:06:01.951885
rcv ctrl pkt info (^p): 1,16,6,3,7,7,-15
splitted in:  [1, 16, 6, 3, 7, 7, -15]
(dst=1 type=0x10(DATA) src=6 seq=3 len=7 SNR=7 RSSI=-15)
post downlink: receive from 6 with pending request
in unicast mode
post downlink: downlink data is "reply from gw"
post downlink: generate downlink/downlink.txt from entry
{"status":"send_request","dst":6,"data":"reply from gw"}
rcv ctrl radio info (^r): 125,5,12
splitted in:  [125, 5, 12]
(BW=125 CR=5 SF=12)
rcv timestamp (^t): 2018-10-05T11:06:01.951
got first framing byte
--> got data prefix
hello
.... many things else here i cut almost ..............
status = send_request
dst = 6
data = reply from gw
--> CAD duration 202
OK1
--> RSSI -128
Packet sent, state 0
JSON record: {"status":"sent","dst":6,"MIC3":"0x07","MIC2":"0xad","MIC1":"0xfc","MIC0":"0x32","data":"reply from gw"}
----------------------------------------------------------

But device did receive nothing.

Please give me some advice to fix them correctly.
Many Thanks.

CongducPham, 2018-10-05:
For a receiver to be able to receive downlink data, it must open a receive window. See the Arduino_LoRa_temp example where a receive window is opened at line 813: https://github.com/CongducPham/LowCostLoRaGw/blob/6721aae1b22be2d8bee89d642481ade700c6aab6/Arduino/Arduino_LoRa_temp/Arduino_LoRa_temp.ino#L813

TienPham77777, 2018-10-05:
Hi,
Thanks for your help!

I did following your but End device still did not receive.
I tried many times to change DELAY_BEFORE_RCVW from 4000 to 15000.
The gateway responded the same before.
And it's a message from End device, one of them:

> SX1276 detected, starting
SX1276 LF/HF calibration
...
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
Reading 0
Temp is -50.00
Sending \!#1#TC/-50.00
Real payload size is 14
\!#1#TC/-50.00
plain payload hex
05 06 07 08 5C 21 23 31 23 54 43 2F 2D 35 30 2E 30 30 
Encrypting
encrypted payload
16 0E C3 72 08 D4 A2 F9 90 80 7B BC 75 D9 78 DF 9F 35 
calculate MIC with NwkSKey
transmitted LoRaWAN-like packet:
MHDR[1] | DevAddr[4] | FCtrl[1] | FCnt[2] | FPort[1] | EncryptedPayload | MIC[4]
40 06 00 00 00 00 00 00 01 16 0E C3 72 08 D4 A2 F9 90 80 7B BC 75 D9 78 DF 9F 35 3B 70 19 75 
end-device uses encapsulated LoRaWAN packet format only for encryption
--> CS1
--> CAD 548
OK1
LoRa pkt size 31
LoRa pkt seq 0
LoRa Sent in 2062
LoRa Sent w/CAD in 2610
Packet sent, state 0
Wait for 8000
Wait for incoming packet
No packet

By default config at source file: [https://github.com/CongducPham/LowCostLoRaGw/blob/6721aae1b22be2d8bee89d642481ade700c6aab6/Arduino/Arduino_LoRa_temp/Arduino_LoRa_temp.ino](https://github.com/CongducPham/LowCostLoRaGw/blob/6721aae1b22be2d8bee89d642481ade700c6aab6/Arduino/Arduino_LoRa_temp/Arduino_LoRa_temp.ino)

> #define WITH_EEPROM
#define WITH_APPKEY
//if you are low on program memory, comment STRING_LIB to save about 2K
//#define STRING_LIB
#define LOW_POWER
#define LOW_POWER_HIBERNATE
#define WITH_AES
//#define LORAWAN
//#define TO_LORAWAN_GW
//#define WITH_ACK
//this will enable a receive window after every transmission
#define WITH_RCVW

How do change it? Please show me the way.
Thank you.
Best Regards.

CongducPham, 2018-10-05:
When you have downlink message from the gateway, make sure that the gateway has correct PABOOST setting for its LoRa radio module. If it is an RFM95-based radio then make sure that PABOOST is defined in radio.makefile on the gateway, then recompile.

TienPham77777, 2018-10-05:
Hi @CongducPham ,

I sure that i config PABOOST all of both, in End device: Arduino_LoRa_temp.ino and Radio.makefile at raspbberry pi gateway. I did compile by command: make raspberry_pi2_downlink.
I use RA-02 LoRa module, change all to 433MHz band (gateway and end device),....

Connection between RA-02 module and gateway (or end device) i use only SPI pins (ss, clk, mosi, miso), other pins were not connected. Is that true?

If it is correct thing, i will try do it again and tell you later.

Best Regards!

CongducPham, 2018-10-05:
We haven't these modules, so I don't know if PA_BOOST is needed ot not. As the SNR at reception is quite good it seems that PABOOST is set correctly. You should not increase DELAY_BEFORE_RCVW because as you see:

> Wait for 8000
> Wait for incoming packet
> No packet

It means that your end-device waits for 8s then open a receive window, it is too late.

regards,

TienPham77777, 2018-10-05:
Because of default value 5000, end device did not receive anything, i increased to some values, etc 6000, 7000,.... but nothing changed. I did not do decrease this value, so maybe i will try later.

I will do create new test to refresh all things to default, then i will notice you.

Best Regards!

TienPham77777, 2018-10-08:
Hi @CongducPham ,

I've just tried to test with a new install (everything's default).
End device:
> //#define BAND868
//#define BAND900
#define BAND433
#define BAND433

> #define PABOOST

> #define WITH_EEPROM
#define WITH_APPKEY
//if you are low on program memory, comment STRING_LIB to save about 2K
//#define STRING_LIB
#define LOW_POWER
#define LOW_POWER_HIBERNATE
#define WITH_AES
//#define LORAWAN
//#define TO_LORAWAN_GW
//#define WITH_ACK
//this will enable a receive window after every transmission
#define WITH_RCVW

Gateway:

> Mode | 1 
Frequency | 433.3
PA_BOOST | Enabled
AES | true 
downlink | 10
wappkey | true

Then i run command: make lora_gateway_pi2_downlink, gateway complied a new lora_gateway replace old file.

End device send data to gateway:

> SX1276 detected, starting
SX1276 LF/HF calibration
...
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
Reading 0
Temp is -50.00
Sending \!#1#TC/-50.00
Real payload size is 14
\!#1#TC/-50.00
plain payload hex
05 06 07 08 5C 21 23 31 23 54 43 2F 2D 35 30 2E 30 30 
Encrypting
encrypted payload
16 0E C3 72 08 D4 A2 F9 90 80 7B BC 75 D9 78 DF 9F 35 
calculate MIC with NwkSKey
transmitted LoRaWAN-like packet:
MHDR[1] | DevAddr[4] | FCtrl[1] | FCnt[2] | FPort[1] | EncryptedPayload | MIC[4]
40 06 00 00 00 00 00 00 01 16 0E C3 72 08 D4 A2 F9 90 80 7B BC 75 D9 78 DF 9F 35 3B 70 19 75 
end-device uses encapsulated LoRaWAN packet format only for encryption
--> CS1
--> CAD 548
OK1
LoRa pkt size 31
LoRa pkt seq 0
LoRa Sent in 2062
LoRa Sent w/CAD in 2610
Packet sent, state 0
Wait for 4000
Wait for incoming packet
No packet
Switch to power saving mode
Successfully switch LoRa module in sleep mode

Gateway responded that has a received and send downlink:

> status = send_request
dst = 6
data = reply from gw
--> CAD duration 201
OK1
--> RSSI -124
Packet sent, state 0
JSON record: {"status":"sent","dst":6,"MIC3":"0x07","MIC2":"0xad","MIC1":"0xfc","MIC0":"0x32","data":"reply from gw"}

I sure that end device couldn't receive anything.
Please give me some advice or could you try to test this issue.

Many thanks,
Best Regards!

CongducPham, 2018-10-08:
Hello, did you generate the downlink request at gw well before data from end-device is received? Because the gateway need to know/read the downlink request before receiving the data from end-device in order to know whether the end-device has pending downlink request or not. So, make sure that there is a downlink request and that gw has read the downlink request, then power on your end-device.

TienPham77777, 2018-10-08:
Hello,

I think i did right by below command (i logged on the gateway in another terminal, then go to downlink path by cd lora_gateway/downlink):

> echo "{\"status\":\"send_request\",\"dst\":6,\"data\":\"reply from gw\"}" > downlink-post.txt

After i use:

> sudo nano downlink-post-queued.txt

This file contain:

> {"status":"send_request","dst":6,"data":"reply from gw"}

Last, i turned on end-device, but string "No packet" still displayed in serial port terminal (COM port that connected to end-device).
What did i do correctly?

CongducPham, 2018-10-08:
Yes, but you have to wait until the gateway reads the downlink-post.txt file. As you set downlink=10 in gateway_conf.json, it means that every 10s, the gateway will read for new downlink request. So wait at least 10s and look that the gateway reads the file before powering on the device.

CongducPham, 2018-10-08:
OK, just to make sure, you are also using an RA-02 LoRa module for the gateway?

TienPham77777, 2018-10-08:
Hi @CongducPham ,

Yes I am, sir :D.

CongducPham, 2018-10-08:
So I really don't see why it is not working right now. Let me try to reproduce the scenario and I'll get back to you. regards,

TienPham77777, 2018-10-08:
I am really looking forward to hearing a good news from you.
Thank you so much!

CongducPham, 2018-10-17:
Hi, I've checked and I found the error that was introduced after adding packet type checking in SX1272.cpp file. It has been corrected. Please get the new SX1272.cpp file for Arduino (not the gateway), recompile your end-device and it should work.

regards,

TienPham77777, 2018-10-19:
Hi Mr. @CongducPham ,

It's glad to hear from you, so i will do it.
Thank you very much.

Best regards!

CongducPham, 2018-10-22:
Great!

KhangNguyen211, 2019-05-20:
Hi @TienPham77777 , I also have a problem like you. I have recompiled my end-device with new libraries SX1272.cpp but  end-device can't receive data by gateway. Can you help me how to solve this problem. Thanks you very much!