CongducPham/LowCostLoRaGw #232: Raspberry Pi Node support?

Wingjam, 2018-10-15:
Do you plan to support Raspberry Pi Node?

That would be nice to have a Raspberry Pi Node communicating with a Raspberry Pi Gateway.

Thank you for that amazing Raspberry Pi Gateway!

CongducPham, 2018-10-16:
What do you mean by PI Node? A device built with a Raspberry? If so, you probably can program such a device as our lib runs on the RPI.

regards,

Wingjam, 2018-10-16:
Hi,
Currently there are only samples for Arduino (Node) communicating with a Raspberry Pi (Gateway). 
Are there any chance to have samples of Raspberry Pi (node) communicating with a Raspberry Pi (Gateway). 

Thank you for your time.

CongducPham, 2018-10-29:
OK, no, we do not plan to provide examples using the PI as end-devices. However, since the SX1272.cpp lib runs on top of the RPI on top of arduPI, you can use the same layer to program your RPI as an arduino node. Hope that helps.

surr2025, 2019-06-26:
Would you please provide ping pong code of Arduino on Raspberry. It will be helpful. 
I tried the ping code (Raspberry Pi as a node). Error: Could not open the serial port: null (arduPi.cpp)

CongducPham, 2019-06-27:
In the ping-pong code, you have to remove everything that is not related to Arduino boards. Take example on the lora_gateway.cpp program that runs on the RPI.

surr2025, 2019-07-07:
I already tried it similar to lora_gateway.cpp. it was throwing an error.

`Unable to open the serial port (null) - `

All ports are open.
I used the spi connection to connect tx-node.