CongducPham/LowCostLoRaGw #234: Not support esp8266 as esp12f

chijiao, 2018-11-01:
I had tested. not support, do you have any plan to support ?

CongducPham, 2018-11-02:
Can you be more specific? We did some tests with ESP8286 and it worked, but there is only minimal support: the SX1272 lib is running fine but all the low-power features are not optimized.

chijiao, 2018-11-04:
Unrecognized transceiver, then print .....  .... ...

vdbpeter, 2020-01-08:
Hi Congduc, I'm also attempting to use the ESP8266 as a node with the RFM95W and having little luck. Fault finding my way through your code, then through the SX1272.cpp & .h, I find the ESP appears to be unable to write/read to the radio. I'm sure the SPI bus is working (tested it with another device), but the setups are coming back with "state 1". With debug on, I saw registers were attempted to be written to, but reading back the registers showing the registers haven't updated.

I severely cut down one of your Arduino_Demo sketches to narrow down the problem:
// Include the SX1272
#include "SX1272.h"

void setup()
{
int e;

  delay(3000);
  // Open serial communications and wait for port to open:
  Serial.begin(115200);  

    
// Select frequency channel
Serial.print(SX1272_SS);
e = sx1272.setChannel(CH_05_900);
Serial.print(F("Setting Channel: state "));
Serial.println(e, DEC);
Serial.println("Should be: 0xE47851, 913.88MHz");
Serial.println(CH_05_900, HEX);

}

void loop() {
  // put your main code here, to run repeatedly:

}

RESULT:
Starting 'setChannel'
 Reading:  ##	Register 1:  0
 Writing:  ##	Register 1:  1
Writing:  ##	Register 6:  E4
 Writing:  ##	Register 7:  78
 Writing:  ##	Register 8:  51
 Reading:  ##	Register 6:  0
 Reading:  ##	Register 7:  0
 Reading:  ##	Register 8:  0
 Writing:  ##	Register 1:  0
Setting Channel: state 1
Should be: 0xE47851, 913.88MHz
E47851

You say you've had an ESP8266 work, so once again, can you please give me some tips to troubleshoot. I think it's quite low level.

Thanks

vdbpeter, 2020-01-08:
Hi- an update: I stumbled on a comment in README (https://github.com/CongducPham/LowCostLoRaGw/tree/master/Arduino)

"Expressif ESP8266 (only the Arduino_LoRa_Simple_temp has been modified to support the ESP8266)"

Trying this sketch with debug on, I found that the setup appears to work. I'll go through the code carefully to find why this works.

starting 'setChannel'
## Reading:  ##	Register 1:  81
## Writing:  ##	Register 1:  81
## Writing:  ##	Register 6:  E4
## Writing:  ##	Register 7:  78
## Writing:  ##	Register 8:  51
## Reading:  ##	Register 6:  E4
## Reading:  ##	Register 7:  78
## Reading:  ##	Register 8:  51
## Frequency channel E47851 has been successfully set ##

We're making progress!
Thanks for your help and guidance.

vdbpeter, 2020-01-08:
Hi, I found this in the code:
#elif defined ARDUINO_ESP8266_ESP01 || defined ARDUINO_ESP8266_NODEMCU
          //in microseconds
          //it is reported that RST pin should be connected to pin 16 to actually reset the board when deepsleep
          //timer is triggered
          ESP.deepSleep(LOW_POWER_PERIOD*1000*1000);

That's needed to wake the thing up, and have a reliable node...can I suggest you add this critical piece of information to your README where you describe ESP8266 support.

Many thanks

CongducPham, 2020-01-09:
Hi, yes, I will add this in the Arduino's README. Thanks for the suggestion. BTW, I'm facing some issue to make my ESP8266 working again, seems that some update on the Arduino IDE created some issue with serial output with the ESP8266, I did not have such issue before with older version of Arduino IDE. Will try to solve this issue in order to flash my ESP8266 again. However, I confirm that it was working fine and LoRa packets were sent correctly.

vdbpeter, 2020-01-09:
Hi, my Arduino version is 1.8.10 and programs my Wemos NodeMCU 0.9 just nicely. My preferred platform is Micropython and this is my first foray with Arduino- and after sorting out which sketch to use and the wiring up of the RST to GPIO15, it runs like clockwork. I shortened the cycle to 1 min, and it hasnt missed a beat for 24 hours. I'm still concerned with the RPi GW resets, but I'll update that Issue.