CongducPham/LowCostLoRaGw #235: Access and modify the Gateway From Internet

matiassantiagotimi, 2018-11-15:
Hi
I would like to know how to access to the gateway "without connectivity with the access point "
For example: I'm far away (200Km) from the gateway and there is a problem. I can't access to "192.168.200.1" because I'm not connected to the access point of the gateway. The only thing Ican see through the internet is the information from the server like "ThingSpeak", etc.
Is there any solution?
Thanks

CongducPham, 2018-11-15:
You can use ngrok or remot3it, see the corresponding README files
- https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-remote-access-ngrok.md
- https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-remote-access-remot3it.md

matiassantiagotimi, 2018-11-15:
Thanks CongducPham !!!