CongducPham/LowCostLoRaGw #236: Development Questions

Shinichi28, 2018-11-22:
Dear Prof. Congduc Pham

I'm very glad to find about your research in Low Cost LoRa Gateway. I'm Roni from Indonesia. I'm Post Graduate student in Politeknik Negeri Semarang. I'm very interesting about your research and i'd like to develop your Low Cost LoRa Gateway to be my Master Thesis Topic, but i don't know what should I do to develop it. My bachelor research is also about DIY Low Cost LoRa Gateway using NodeMCU and MQTT protocol. I'm very glad if you want to reply this message to my email address, roni.apriantoro28@gmail.com.

Best Regards, 

Roni

CongducPham, 2018-11-22:
Hi, thank you for your interest. Happy to know that you want to develop on our framework. There are a couple if things that we like to develop such:

- better support of sensor management with a dedicated web server on the gateway to extract and visualize various parameters from sensors that are connected to the gateway (by parsing the MongoDB or the post-processing.log file for instance
- make the gateway able to connect to other clouds platforms such as TheThingNetwork, Dropbox, GoogleDrive, AWS, ....

regards,

Shinichi28, 2018-11-25:
Hi Prof,
Thanks for your fast respond, and forgive me for my late respond. I'll discuss it with my lecturer for point number one. I'm very glad if my lecturer agree to develop a low cost lora gateway with your framework.

regards,