CongducPham/LowCostLoRaGw #237: LoRa communication between node and 8-channel gateway

hibish123, 2019-01-18:
Hi,

Thank you so much for the script. Executing `./lora_gateway_pi2` worked very well using Raspberry Pi 3B+ and LoRa GPS Hat as gateway. However, the gateway I used was single channel gateway. Now, I would like to use 8-channel gateway. Before I proceed with the purchase of an 8-channel gateway, I would like to ask which 8-channel gateway would you suggest would work well with the code? Some of the gateway suggestions I have in mind are RPi 3 + iC880a OR RPi 3 + RaK831. Please let me know your suggestion.

CongducPham, 2019-01-19:
Our framwork is not for 8-channel radio concentrator. For that you should use lorawan cod for instance, or code from the thing networks. regards,

hibish123, 2019-01-19:
Thank you so much Dr. I just have one last question. Which Lorawan code are you referring to?

sjaimohd, 2019-05-08:
Hi @hibish123,

Do you manage to change to 8 channel?