CongducPham/LowCostLoRaGw #238: Update Git version: 379 - Erro Cloud

viniaero, 2019-02-10:
Hi, I update software for version 379, I noticed that stopped sending to cloud, the gateway receives more package does not send to cloud.

I installed software in another micro sd, when putting data (example:
ThingSpeak) I enter the data dou apply, I give F5 the page appears without the data ..

GATEWAY
2019-02-10T23:28:00.819118> got first framing byte
2019-02-10T23:28:00.820530> --> got LoRa data prefix
2019-02-10T23:28:00.821677> --> DATA with_appkey: read app key sequence
2019-02-10T23:28:00.827366> app key is  0x0a 0x23 0x23 0x20
2019-02-10T23:28:00.830406> app key disabled
2019-02-10T23:28:00.832990> ACK set and written in FIFO ##
2019-02-10T23:28:00.838391> ## ACK to send:
2019-02-10T23:28:00.839834> Destination: 3
2019-02-10T23:28:00.840954> Source: 1
2019-02-10T23:28:00.842052> ACK number: 46
2019-02-10T23:28:00.843205> ACK length: 2
2019-02-10T23:28:00.844514> ACK payload: 0
2019-02-10T23:28:00.845652> ACK SNR last rcv pkt: 6
2019-02-10T23:28:00.854967>  ##
2019-02-10T23:28:00.856136> 
2019-02-10T23:28:00.857225> --> CS1
2019-02-10T23:28:00.858348> --> CAD 202
2019-02-10T23:28:00.861686> OK1
2019-02-10T23:28:00.862834> --> RSSI -126
2019-02-10T23:28:00.863942> ACK requested by 3
2019-02-10T23:28:00.865052> --- rxlora. dst=1 type=0x1A src=3 seq=46 len=0 SNR=6 RSSIpkt=-39 BW=125 CR=4/5 SF=12
2019-02-10T23:28:00.866231> 2019-02-10T23:28:00.853766
2019-02-10T23:28:00.867416> rcv ctrl pkt info (^p): 1,26,3,46,0,6,-39
2019-02-10T23:28:00.868705> splitted in:  [1, 26, 3, 46, 0, 6, -39]
2019-02-10T23:28:00.870157> (dst=1 type=0x1A(DATA WAPPKEY ACK_REQ) src=3 seq=46 len=0 SNR=6 RSSI=-39)
2019-02-10T23:28:00.871303> rcv ctrl radio info (^r): 125,5,12
2019-02-10T23:28:00.872435> splitted in:  [125, 5, 12]
2019-02-10T23:28:00.873543> (BW=125 CR=5 SF=12)
2019-02-10T23:28:00.874630> rcv timestamp (^t): 2019-02-10T23:28:00.826
2019-02-10T23:28:00.875800

NODE:
Reading 68
Reading 70
Reading 69
Reading 68
Mean temp is 33.59
Sending \!#3#TC/33.59
Real payload size is 13
--> CS1
--> CAD 545
OK1
wait for ACK

Starting 'getACK'
## ACK received:
Destination: 3
Source: 1
ACK number: 46
ACK length: 2
ACK payload: 0
ACK SNR of rcv pkt at gw: 6
##

RetryLoRa pkt size 17
LoRa pkt seq 46
LoRa Sent in 3593
LoRa Sent w/CAD in 4138
Packet sent, state 0
Remaining ToA is 36000
Switch to power saving mode
Successfully switch LoRa module in sleep mode
...........................

CongducPham, 2019-02-11:
Hi, there is probably 2 errors. The first one I see is that you are sending !#3#TC/33.59. If you want to upload to cloud then the prefix should be \!. If you are using our example template then I don't know why the \ is missing. Please check.

Then maybe the second error. Have you perform basic config after updating?

viniaero, 2019-02-11:
Yes after installation, I applied update of config basic.

Yes I am using the simple sensor example. just changed the band and channel.

the strange one that I enter the written key thingspeak, I insert the id sensor and I put True, more when I leave the page and I return this everything zeroed (erased)

CongducPham, 2019-02-12:
Not sure to understand, please copy/paste the portion of code you are mentioning.

viniaero, 2019-02-13:
Thanks for the help, (For some reason the copy and paste text, does not appear \) it follows print of the screen

[image] 

I put the key and updated the screen (F5) back all white
[image]

Thank you for your help

CongducPham, 2019-02-13:
You probably need to update the web admin interface. Use the option in gateway configuration menu.

viniaero, 2019-02-13:
I have already updated the interface, I am using the raspberry zero gateway pi, could be the problem?

Thank you

closed 2019-02-26 by viniaero