CongducPham/LowCostLoRaGw #239: arduino lora to arduino gateway won't print the package content

ucod, 2019-02-15:
I'm running arduino_simple_temp & arduino_lora_gateway on two esp32 with a sx1276. Everything is default, except eeprom and lowpower are commented out because they won't compile on esp32 platform. 

The package is being sent
> Reading 836
Reading 761
Reading 704
Reading 656
Mean temp is 311.26
Sending \!#1#TC/311.26
Real payload size is 14
--> CS1
       --> CAD 548
OK1
   --> RSSI -124
LoRa pkt size 14
LoRa pkt seq 0
LoRa Sent in 1570
LoRa Sent w/CAD in 2119
Packet sent, state 0

On the receive terminal, I get
>^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
^$Low-level gw status ON
--- rxlora. dst=1 type=0x12 src=6 seq=0 len=0 SNR=6 RSSIpkt=-39 BW=125 CR=4/5 SF=12
^p1,18,6,0,0,6,-39
^r125,5,12
>       
>XIYC3Z3BJjX{(bs+hs+H,c,cA,C	SX1276 detected, starting
>SX1276 LF/HF calibration
>...
                         
And the content of the package is not printed. 
As for received hex,
>00000220 75 73 20 4F 4E 0D 0A 2D   2D 2D 20 72 78 6C 6F 72  us ON..--- rxlor
00000230 61 2E 20 64 73 74 3D 31   20 74 79 70 65 3D 30 78  a. dst=1 type=0x
00000240 31 32 20 73 72 63 3D 36   20 73 65 71 3D 30 20 6C  12 src=6 seq=0 l
00000250 65 6E 3D 30 20 53 4E 52   3D 36 20 52 53 53 49 70  en=0 SNR=6 RSSIp
00000260 6B 74 3D 2D 33 39 20 42   57 3D 31 32 35 20 43 52  kt=-39 BW=125 CR
00000270 3D 34 2F 35 20 53 46 3D   31 32 0A 5E 70 31 2C 31  =4/5 SF=12.^p1,1
00000280 38 2C 36 2C 30 2C 30 2C   36 2C 2D 33 39 0A 5E 72  8,6,0,0,6,-39.^r
00000290 31 32 35 2C 35 2C 31 32   0A FF FE 0D 0A 58 EC B0  125,5,12.....X..
000002a0 90 0F 49 D8 11 C3 DD 59   43 17 33 5A 33 82 B3 10  ..I....YC.3Z3...
000002b0 42 12 4A 13 0B A3 13 9B   6A C4 DC 91 92 DB 13 D2  B.J.....j.......
000002c0 91 C0 0F DD 58 03 F2 B1   D0 C9 D2 81 F0 7B D2 81  ....X........{..

The letters after 12 are init garbage/noise from the reset input button. If i let it run, I get nothing.

CongducPham, 2019-02-15:
Have you also get the latest version of the SX1272.cpp lib?

ucod, 2019-02-15:
yes, i've cloned this repo ~2 days ago

viniaero, 2019-02-26:
Hi, did you solve it? Yesterday an update happened, but now I do not receive anymore package at the gateway, both the  Arduino_LoRa_Ping_Pong_LCD example and Arduino_LoRa_Simple_temp

CongducPham, 2019-02-26:
I've updated the SX1272.* lib. So please take the new version to compile for your ESP32 devices. There is also the Arduino_LoRa_Gateway.ino sketch that has been updated. Maybe you already got the right version but just in case, get the new one and retry your setup.

viniaero, 2019-02-26:
I upgraded the Sx1272 library and the gateway to 382, ​​continuing the gateway not receiving packet. I am using Arduino uno for example Simple_temp and Ping Pong Lcd arduino nano.

viniaero, 2019-02-26:
Solved, thank you....

CongducPham, 2019-02-26:
Great!
regards,

closed 2020-06-11 by ucod