CongducPham/LowCostLoRaGw #240: Gateway always receives unknown signs

niceboat15, 2019-02-26:
Hello Dr,

I am using RPI3 with RFM95W module as my gateway, my end device is an Arduino unoR3 and a BME280 with the another RFM95W module.

I set up the gateway as instructed, and I used Arduino_LoRa_Ping_Pong to test it. It worked well on end-device side, it can send ping and receive pong.

At the gateway side, every packet was receive correctly, but there is no ping message display on the terminal. All messages were replaced by "??". No matter what data I want to send to the gateway, the gateway will only receive "??".

Could you help me with this issue? Very appreciate your help.

[gateway_log.txt](https://github.com/CongducPham/LowCostLoRaGw/files/2906084/gateway_log.txt)
[image]

Best regards !

niceboat15, 2019-02-26:
I just update the gateway software from 379 to 382 and it works
The message comes with "??Ping"
How this happen just in 379 version?

CongducPham, 2019-02-26:
Because I changed recently how the packet length is obtained. So both SX1272.cpp and lora_gateway.cpp must be updated. If only one of those is updated, there is the error where len=0 as you can see in your screenshoot.
regards,

ucod, 2019-02-26:
"??" is an empty line. for some funny reason the arduino console prints them mirrored ⸮, and I have had the same issue [here](https://github.com/CongducPham/LowCostLoRaGw/issues/239), except _moserial_ doesn't print question marks.

CongducPham, 2019-02-26:
The ?? actually are non-printable characters. There are 0xFF and 0xFE that are the prefix sent by the low-level gateway for the post-processing stage to indicate LoRa data reception.

regards,

ucod, 2019-02-26:
Oh, ty for the info.