CongducPham/LowCostLoRaGw #241: Gateway isn't receiving packets after Basic config or install the latest version

avanjur, 2019-03-10:
Dear Pham

I have a strange problem.

The gateway is rpi 3 B+ with generic Semtech 1276 radio module from ebay (https://www.ebay.com/itm/868-915MHz-SX1276-Wireless-Transceiver-Module-LoRa-SPI-remote-alarm-/302789176919?var=&hash=item0)
I have downloaded the latest img of the gateway (version 372 ), and flashed on 32gb micro SD card.

The node is a pro mini with the same 1276 radio module flashed with untouched Arduino_LoRa_Generic_Simple_MultiSensors code.

If I power on the gateway and the node, the packets are arriving on the gateway, everything is working fine.

If I leave the setting alone, and just upgrade the gateway to the latest version ( 382 ) via web admin, the gateway is not receiving packets anymore.

After I do a fresh install on the gateway (372), and is working ok.
Setup the Radio module in web admin to: Mode 1, ISM Band 868, Frequency 865.2, PA_BOOST enabled (i tried disabling to), and do Gateway update > Basic config, restart the gateway.
The gateway is not receiving packets.

After I do a fresh install on the gateway (372), and is working ok.
I leave the setting alone, and just do Gateway update > Basic config and restart the gateway.
The gateway is not receiving packets.
...

Finally a have find out, i can enable/disable Nodered, AES, app key, and the gateway is working fine after update, but if i touch update or basic config the gateway stop receiving packets.

After many reties i am very confused, what i miss ?

I have pasted a log when is working fine, and a log when is not.

Log file when is working :
Sun 10 Mar 11:07:39 UTC 2019
2019-03-10T12:07:23.735922> rcv ctrl pkt info (^p): 1,18,6,0,79,8,-41
2019-03-10T12:07:23.736100> splitted in:  [1, 18, 6, 0, 79, 8, -41]
2019-03-10T12:07:23.736256> (dst=1 type=0x12(DATA WAPPKEY) src=6 seq=0 len=79 SNR=8 RSSI=-41)
2019-03-10T12:07:23.736427> rcv ctrl radio info (^r): 125,5,12
2019-03-10T12:07:23.736632> splitted in:  [125, 5, 12]
2019-03-10T12:07:23.736796> (BW=125 CR=5 SF=12)
2019-03-10T12:07:23.736960> rcv timestamp (^t): 2019-03-10T12:07:23.734
2019-03-10T12:07:23.737231>
2019-03-10T12:07:23.737427> got first framing byte
2019-03-10T12:07:23.737653> --> got LoRa data prefix
2019-03-10T12:07:23.737811> --> DATA with_appkey: read app key sequence
2019-03-10T12:07:23.738004> app key is  0x05 0x06 0x07 0x08
2019-03-10T12:07:23.738182> app key disabled
2019-03-10T12:07:23.738339> valid app key: accept data
2019-03-10T12:07:23.738542> number of enabled clouds is 1
2019-03-10T12:07:23.738757> --> cloud[0]
2019-03-10T12:07:23.738909> uploading with python CloudThingSpeak.py
2019-03-10T12:07:23.739081> python CloudThingSpeak.py "LM35/124.84/TMP36/94.69/TC1/-1.0/HU1/-1.0/TC2/-100.0/HU2/-100.0/DS/-127.0" "1,18,6,0,79,8,-41" "125,5,12" "2019-03-10T12:07:23+01:00" "00000027EBE6BE13"
2019-03-10T12:07:25.147148> ThingSpeak: uploading (multiple)
2019-03-10T12:07:25.147913> rcv msg to log (\!) on ThingSpeak ( default , default ):
2019-03-10T12:07:25.148380> ThingSpeak: will issue curl cmd
2019-03-10T12:07:25.148825> curl -s -k -X POST --data field1=124.84&field2=94.69&field3=-1.0&field4=-1.0&field5=-100.0&field6=-100.0&field7=-127.0 https://api.thingspeak.com/update?key=SGSH52UGPVAUYG3S
2019-03-10T12:07:25.149273> ThingSpeak: returned code from server is 457475
2019-03-10T12:07:25.153682> --> cloud end
2019-03-10T12:07:25.154177>
2019-03-10T12:07:45.850196> --- rxlora. dst=1 type=0x12 src=6 seq=0 len=79 SNR=9 RSSIpkt=-43 BW=125 CR=4/5 SF=12
2019-03-10T12:07:45.850671> 2019-03-10T12:07:45.849959
2019-03-10T12:07:45.850914> rcv ctrl pkt info (^p): 1,18,6,0,79,9,-43
2019-03-10T12:07:45.851094> splitted in:  [1, 18, 6, 0, 79, 9, -43]
2019-03-10T12:07:45.851262> (dst=1 type=0x12(DATA WAPPKEY) src=6 seq=0 len=79 SNR=9 RSSI=-43)
2019-03-10T12:07:45.851460> rcv ctrl radio info (^r): 125,5,12
2019-03-10T12:07:45.851652> splitted in:  [125, 5, 12]
2019-03-10T12:07:45.851825> (BW=125 CR=5 SF=12)
2019-03-10T12:07:45.852049> rcv timestamp (^t): 2019-03-10T12:07:45.849
2019-03-10T12:07:45.852206>
2019-03-10T12:07:45.852401> got first framing byte
2019-03-10T12:07:45.852597> --> got LoRa data prefix
2019-03-10T12:07:45.852795> --> DATA with_appkey: read app key sequence
2019-03-10T12:07:45.852970> app key is  0x05 0x06 0x07 0x08
2019-03-10T12:07:45.853128> app key disabled
[... 67 lines cut ...]
Log file when is not working :
2019-03-10T10:50:51.118463> post_processing_gw.py found an alert_conf section
2019-03-10T10:50:51.119165> Parsing cloud declarations
2019-03-10T10:50:51.119747> [u'python CloudNodeRed.py']
2019-03-10T10:50:51.120611> Parsed all cloud declarations
2019-03-10T10:50:51.121539> post_processing_gw.py got cloud list:
2019-03-10T10:50:51.122203> [u'python CloudNodeRed.py']
2019-03-10T10:50:51.122763> Parsing cloud declarations
2019-03-10T10:50:51.123364> Parsed all cloud declarations
2019-03-10T10:50:51.124034> post_processing_gw.py got encrypted cloud list:
2019-03-10T10:50:51.128248> []
2019-03-10T10:50:51.128853> Parsing cloud declarations
2019-03-10T10:50:51.129486> Parsed all cloud declarations
2019-03-10T10:50:51.130071> post_processing_gw.py got LoRaWAN encrypted cloud list:
2019-03-10T10:50:51.130702> []
2019-03-10T10:50:51.131282> Starting thread to perform periodic gw status/tasks
2019-03-10T10:50:51.131853> 2019-03-10 10:50:51.118751
2019-03-10T10:50:51.132354> post status: gw ON
2019-03-10T10:50:51.132881> post status: executing periodic tasks
2019-03-10T10:50:51.272983> post status: start running
2019-03-10T10:50:51.274509> post status: show current GPS position
2019-03-10T10:50:51.278674> post status show GPS: current GPS coordinate: gw lat my_lat long my_long
2019-03-10T10:50:51.279541> post status: exiting
2019-03-10T10:50:52.122298>
2019-03-10T10:50:52.122677> Current working directory: /home/pi/lora_gateway

CongducPham, 2019-03-10:
Hi, you say that you have a RPI3B+ right? Maybe this is the issue. I haven't updated the script to detect the 3B+ so when you do an update it compile for the RPI1. With fresh install, then the executable is for the RPI3B which is compatible with the RPI3B+. So to try to solve the issue, log on the RPI and do manually: make lora_gateway_pi2, then restart. Should work then.

I'm updating the scripts on the github so that the RPI3B+ is correcty detected. Fortunatly, using the 3B+ is not that common right now and I'm also not recommending it because it consumes more power, unless you really need the extra processing power, which is not the case if tou only use it as a gateway.

regards,

FrancoTron, 2019-03-11:
Hi, I have the same problem. But, when I initialize the RPI3 just it set for RPI1.
I use rpi3 since in my country I only get this version.

CongducPham, 2019-03-11:
Is it the RPI3B or RPI3B+? Because if my guess is correct, you should not have this problem if you have an RPI3B.

regards,

FrancoTron, 2019-03-11:
It is RPI3 B+.
Thanks

avanjur, 2019-03-11:
"So to try to solve the issue, log on the RPI and do manually: make lora_gateway_pi2, then restart. Should work then."

Working like a charm.

Thanks.

closed 2019-03-11 by avanjur