CongducPham/LowCostLoRaGw #244: Post Processing Error

rafd02, 2019-04-24:
Dear CongducPham,

Iam new to GITHUB. 

I use Raspberry Pi 3B
https://www.cytron.io/p-raspberry-pi-3-model-b?search=raspberry%20pi&description=1&src=search

RFM95W-915S2 RF Transceiver Module 915 MHz
https://my.rs-online.com/web/p/rf-modules/1251260/

It is a fresh install and this is the output of my post processing. I did new installation and full update and reboot. This is the output i get. I already referred to the https://github.com/CongducPham/LowCostLoRaGw/issues/241. 
I have done it and it still the same.

The radio i am using is detected by the program. Only the post processing is my problem

BEGIN OUTPUT
Running lora_gateway with post-processing... CTRL-C to exit
post_processing_gw.py found an alert_conf section
Parsing cloud declarations
[u'python CloudThingSpeak.py']
Parsed all cloud declarations
post_processing_gw.py got cloud list: 
[u'python CloudThingSpeak.py']
Parsing cloud declarations
Parsed all cloud declarations
post_processing_gw.py got encrypted cloud list: 
[]
Parsing cloud declarations
Parsed all cloud declarations
post_processing_gw.py got LoRaWAN encrypted cloud list: 
[]
Starting thread to perform periodic gw status/tasks
2019-04-25 03:28:05.827135
post status: gw ON
post status: executing periodic tasks
post status: start running
post status: exiting

Current working directory: /home/pi/lora_gateway

CongducPham, 2019-04-25:
I don't see the error, can you be more specific?

rafd02, 2019-04-25:
Thanks for the reply. I think it should be like this. This is an example from other people output

Log file when is working :
Sun 10 Mar 11:07:39 UTC 2019
2019-03-10T12:07:23.735922> rcv ctrl pkt info (^p): 1,18,6,0,79,8,-41
2019-03-10T12:07:23.736100> splitted in: [1, 18, 6, 0, 79, 8, -41]
2019-03-10T12:07:23.736256> (dst=1 type=0x12(DATA WAPPKEY) src=6 seq=0 len=79 SNR=8 RSSI=-41)
2019-03-10T12:07:23.736427> rcv ctrl radio info (^r): 125,5,12
2019-03-10T12:07:23.736632> splitted in: [125, 5, 12]
2019-03-10T12:07:23.736796> (BW=125 CR=5 SF=12)
2019-03-10T12:07:23.736960> rcv timestamp (^t): 2019-03-10T12:07:23.734
2019-03-10T12:07:23.737231>
2019-03-10T12:07:23.737427> got first framing byte
2019-03-10T12:07:23.737653> --> got LoRa data prefix
2019-03-10T12:07:23.737811> --> DATA with_appkey: read app key sequence
2019-03-10T12:07:23.738004> app key is 0x05 0x06 0x07 0x08
2019-03-10T12:07:23.738182> app key disabled
2019-03-10T12:07:23.738339> valid app key: accept data
2019-03-10T12:07:23.738542> number of enabled clouds is 1
2019-03-10T12:07:23.738757> --> cloud[0]
2019-03-10T12:07:23.738909> uploading with python CloudThingSpeak.py
2019-03-10T12:07:23.739081> python CloudThingSpeak.py "LM35/124.84/TMP36/94.69/TC1/-1.0/HU1/-1.0/TC2/-100.0/HU2/-100.0/DS/-127.0" "1,18,6,0,79,8,-41" "125,5,12" "2019-03-10T12:07:23+01:00" "00000027EBE6BE13"
2019-03-10T12:07:25.147148> ThingSpeak: uploading (multiple)
2019-03-10T12:07:25.147913> rcv msg to log (!) on ThingSpeak ( default , default ):
2019-03-10T12:07:25.148380> ThingSpeak: will issue curl cmd
2019-03-10T12:07:25.148825> curl -s -k -X POST --data field1=124.84&field2=94.69&field3=-1.0&field4=-1.0&field5=-100.0&field6=-100.0&field7=-127.0 https://api.thingspeak.com/update?key=SGSH52UGPVAUYG3S
2019-03-10T12:07:25.149273> ThingSpeak: returned code from server is 457475
2019-03-10T12:07:25.153682> --> cloud end
2019-03-10T12:07:25.154177>
2019-03-10T12:07:45.850196> --- rxlora. dst=1 type=0x12 src=6 seq=0 len=79 SNR=9 RSSIpkt=-43 BW=125 CR=4/5 SF=12
2019-03-10T12:07:45.850671> 2019-03-10T12:07:45.849959
2019-03-10T12:07:45.850914> rcv ctrl pkt info (^p): 1,18,6,0,79,9,-43
2019-03-10T12:07:45.851094> splitted in: [1, 18, 6, 0, 79, 9, -43]
2019-03-10T12:07:45.851262> (dst=1 type=0x12(DATA WAPPKEY) src=6 seq=0 len=79 SNR=9 RSSI=-43)
2019-03-10T12:07:45.851460> rcv ctrl radio info (^r): 125,5,12
2019-03-10T12:07:45.851652> splitted in: [125, 5, 12]
2019-03-10T12:07:45.851825> (BW=125 CR=5 SF=12)
2019-03-10T12:07:45.852049> rcv timestamp (^t): 2019-03-10T12:07:45.849
2019-03-10T12:07:45.852206>
2019-03-10T12:07:45.852401> got first framing byte
2019-03-10T12:07:45.852597> --> got LoRa data prefix
2019-03-10T12:07:45.852795> --> DATA with_appkey: read app key sequence
2019-03-10T12:07:45.852970> app key is 0x05 0x06 0x07 0x08
2019-03-10T12:07:45.853128> app key disabled
2019-03-10T12:07:45.853277> valid app key: accept data
2019-03-10T12:07:45.853465> number of enabled clouds is 1
2019-03-10T12:07:45.853666> --> cloud[0]
2019-03-10T12:07:45.853835> uploading with python CloudThingSpeak.py
2019-03-10T12:07:45.854029> python CloudThingSpeak.py "LM35/152.23/TMP36/91.92/TC1/-1.0/HU1/-1.0/TC2/-100.0/HU2/-100.0/DS/-127.0" "1,18,6,0,79,9,-43" "125,5,12" "2019-03-10T12:07:45+01:00" "00000027EBE6BE13"
2019-03-10T12:07:47.223620> ThingSpeak: uploading (multiple)
2019-03-10T12:07:47.225251> rcv msg to log (!) on ThingSpeak ( default , default ):
2019-03-10T12:07:47.226913> ThingSpeak: will issue curl cmd
2019-03-10T12:07:47.228256> curl -s -k -X POST --data field1=152.23&field2=91.92&field3=-1.0&field4=-1.0&field5=-100.0&field6=-100.0&field7=-127.0 https://api.thingspeak.com/update?key=SGSH52UGPVAUYG3S
2019-03-10T12:07:47.229805> ThingSpeak: returned code from server is 457476
2019-03-10T12:07:47.231013> --> cloud end
2019-03-10T12:07:47.232177>
2019-03-10T12:08:07.965982> --- rxlora. dst=1 type=0x12 src=6 seq=0 len=80 SNR=7 RSSIpkt=-40 BW=125 CR=4/5 SF=12
2019-03-10T12:08:07.967067> 2019-03-10T12:08:07.965744
2019-03-10T12:08:07.968181> rcv ctrl pkt info (^p): 1,18,6,0,80,7,-40
2019-03-10T12:08:07.969241> splitted in: [1, 18, 6, 0, 80, 7, -40]
2019-03-10T12:08:07.969729> (dst=1 type=0x12(DATA WAPPKEY) src=6 seq=0 len=80 SNR=7 RSSI=-40)
2019-03-10T12:08:07.970060> rcv ctrl radio info (^r): 125,5,12
2019-03-10T12:08:07.970382> splitted in: [125, 5, 12]
2019-03-10T12:08:07.971529> (BW=125 CR=5 SF=12)
2019-03-10T12:08:07.972601> rcv timestamp (^t): 2019-03-10T12:08:07.965
2019-03-10T12:08:07.973748>
2019-03-10T12:08:07.974772> got first framing byte
2019-03-10T12:08:07.975840> --> got LoRa data prefix
2019-03-10T12:08:07.976765> --> DATA with_appkey: read app key sequence
2019-03-10T12:08:07.977977> app key is 0x05 0x06 0x07 0x08
[... 15 lines cut ...]
2019-03-10T12:08:30.082059> rcv ctrl pkt info (^p): 1,18,6,0,78,7,-39
2019-03-10T12:08:30.082840> splitted in: [1, 18, 6, 0, 78, 7, -39]
2019-03-10T12:08:30.083479> (dst=1 type=0x12(DATA WAPPKEY) src=6 seq=0 len=78 SNR=7 RSSI=-39)
2019-03-10T12:08:30.084273> rcv ctrl radio info (^r): 125,5,12
2019-03-10T12:08:30.084861> splitted in: [125, 5, 12]
2019-03-10T12:08:30.085675> (BW=125 CR=5 SF=12)
2019-03-10T12:08:30.086212> rcv timestamp (^t): 2019-03-10T12:08:30.079
2019-03-10T12:08:30.087107>
2019-03-10T12:08:30.087688> got first framing byte
2019-03-10T12:08:30.089602> --> got LoRa data prefix
2019-03-10T12:08:30.090068> --> DATA with_appkey: read app key sequence
2019-03-10T12:08:30.090414> app key is 0x05 0x06 0x07 0x08
[... cut ...]
2019-03-10T12:08:31.406593> rcv msg to log (!) on ThingSpeak ( default , default ):
2019-03-10T12:08:31.407324> ThingSpeak: will issue curl cmd
2019-03-10T12:08:31.407920> curl -s -k -X POST --data field1=141.99&field2=45.0&field3=-1.0&field4=-1.0&field5=-100.0&field6=-100.0&field7=-127.0 https://api.thingspeak.com/update?key=SGSH52UGPVAUYG3S
2019-03-10T12:08:31.408532> ThingSpeak: returned code from server is 0, do not retry
2019-03-10T12:08:31.414244> --> cloud end
2019-03-10T12:08:31.414845>

CongducPham, 2019-04-25:
OK, but are you sending something with a device? If yes, what example are you using?

rafd02, 2019-04-25:
I am sending from Arduino_Interactive Device Sketch. There is no receive message from Arduino to my Gateway. I am sorry if my English is not good. I already changed to Channel12_900 to both Gateway and Arduino which is to be specific 915.00 frequency

CongducPham, 2019-04-25:
Did you check that PABOOST is uncommented? Also, try with the Arduino_LoRa_Simple_temp and make sure that BAN900 is selected and that line 174,175 and 176 are as follows:

const uint32_t DEFAULT_CHANNEL=CH_05_900;
//For HongKong, Japan, Malaysia, Singapore, Thailand, Vietnam: 920.36MHz     
//const uint32_t DEFAULT_CHANNEL=CH_08_900;

Then, on the gateway, you have to set the frequency to 913.88

rafd02, 2019-04-26:
Hi, Sorry for the late reply. I have tested it. The output is like this. Does it mean it worked? 

2019-04-26T15:52:04.698365> --- rxlora. dst=1 type=0x12 src=8 seq=8 len=15 SNR=8 RSSIpkt=-53 BW=125 CR=4/5 SF=12
2019-04-26T15:52:04.698825> 2019-04-26T15:52:04.698129
2019-04-26T15:52:04.699135> rcv ctrl pkt info (^p): 1,18,8,8,15,8,-53
2019-04-26T15:52:04.699421> splitted in:  [1, 18, 8, 8, 15, 8, -53]
2019-04-26T15:52:04.699705> (dst=1 type=0x12(DATA WAPPKEY) src=8 seq=8 len=15 SNR=8 RSSI=-53)
2019-04-26T15:52:04.699987> rcv ctrl radio info (^r): 125,5,12,913880
2019-04-26T15:52:04.700264> splitted in:  [125, 5, 12, 913880]
2019-04-26T15:52:04.700522> (BW=125 CR=5 SF=12)
2019-04-26T15:52:04.700806> rcv timestamp (^t): 2019-04-26T15:52:04.697
2019-04-26T15:52:04.701065> 
2019-04-26T15:52:04.701294> adding time zone info
2019-04-26T15:52:04.701525> new rcv timestamp (^t): 2019-04-26T15:52:04.697+08:00
2019-04-26T15:52:04.701793> got first framing byte
2019-04-26T15:52:04.702089> --> got LoRa data prefix
2019-04-26T15:52:04.702445> --> DATA with_appkey: read app key sequence
2019-04-26T15:52:04.702715> app key is  0x05 0x06 0x07 0x08
2019-04-26T15:52:04.702962> app key disabled
2019-04-26T15:52:04.703217> valid app key: accept data
2019-04-26T15:52:04.703512> number of enabled clouds is 1
2019-04-26T15:52:04.703779> --> cloud[0]
2019-04-26T15:52:04.704044> uploading with python CloudThingSpeak.py
2019-04-26T15:52:04.704315> python CloudThingSpeak.py "TC/331.15" "1,18,8,8,11,8,-53" "125,5,12,913880" "2019-04-26T15:52:04.697+08:00" "0000B827EB28D4CB"
2019-04-26T15:52:07.441403> ThingSpeak: uploading (multiple)
2019-04-26T15:52:07.442146> rcv msg to log (\!) on ThingSpeak ( default , default ): 
2019-04-26T15:52:07.442710> ThingSpeak: will issue curl cmd
2019-04-26T15:52:07.443285> curl -s -k -X POST --data field1=331.15 https://api.thingspeak.com/update?key=SGSH52UGPVAUYG3S
2019-04-26T15:52:07.443782> ThingSpeak: returned code from server is 544823
2019-04-26T15:52:07.448440> --> cloud end
2019-04-26T15:52:07.449031> 
2019-04-26T15:52:25.790381> --- rxlora. dst=1 type=0x12 src=8 seq=9 len=15 SNR=9 RSSIpkt=-52 BW=125 CR=4/5 SF=12
2019-04-26T15:52:25.791049> 2019-04-26T15:52:25.790153
2019-04-26T15:52:25.791538> rcv ctrl pkt info (^p): 1,18,8,9,15,9,-52
2019-04-26T15:52:25.792135> splitted in:  [1, 18, 8, 9, 15, 9, -52]
2019-04-26T15:52:25.792596> (dst=1 type=0x12(DATA WAPPKEY) src=8 seq=9 len=15 SNR=9 RSSI=-52)
2019-04-26T15:52:25.793090> rcv ctrl radio info (^r): 125,5,12,913880
2019-04-26T15:52:25.793519> splitted in:  [125, 5, 12, 913880]
2019-04-26T15:52:25.793938> (BW=125 CR=5 SF=12)
2019-04-26T15:52:25.794536> rcv timestamp (^t): 2019-04-26T15:52:25.789
2019-04-26T15:52:25.795033> 
2019-04-26T15:52:25.795486> adding time zone info
2019-04-26T15:52:25.796022> new rcv timestamp (^t): 2019-04-26T15:52:25.789+08:00
2019-04-26T15:52:25.796626> got first framing byte
2019-04-26T15:52:25.797362> --> got LoRa data prefix
2019-04-26T15:52:25.797988> --> DATA with_appkey: read app key sequence
2019-04-26T15:52:25.798786> app key is  0x05 0x06 0x07 0x08
2019-04-26T15:52:25.799301> app key disabled
2019-04-26T15:52:25.799736> valid app key: accept data
2019-04-26T15:52:25.800235> number of enabled clouds is 1
2019-04-26T15:52:25.800766> --> cloud[0]
2019-04-26T15:52:25.807137> uploading with python CloudThingSpeak.py
2019-04-26T15:52:25.807737> python CloudThingSpeak.py "TC/331.45" "1,18,8,9,11,9,-52" "125,5,12,913880" "2019-04-26T15:52:25.789+08:00" "0000B827EB28D4CB"
2019-04-26T15:52:28.205873> ThingSpeak: uploading (multiple)
2019-04-26T15:52:28.206939> rcv msg to log (\!) on ThingSpeak ( default , default ): 
2019-04-26T15:52:28.207750> ThingSpeak: will issue curl cmd
2019-04-26T15:52:28.208446> curl -s -k -X POST --data field1=331.45 https://api.thingspeak.com/update?key=SGSH52UGPVAUYG3S
2019-04-26T15:52:28.209283> ThingSpeak: returned code from server is 544824
2019-04-26T15:52:28.212384> --> cloud end
2019-04-26T15:52:28.213161>

rafd02, 2019-04-26:
THANK YOU ChongducPham!
I don't know how to express my feeling towards you.
I wish you the best for your life. You have been contribute to the community so well. If we can meet in person real life. I want you to know that i am the most happiest person to meet you.

closed 2019-04-26 by rafd02

CongducPham, 2019-04-26:
Great, yes it seems working. You can then go further. Thank you for the kind feedback. I appreciated.

All the best,