CongducPham/LowCostLoRaGw #245: TTN LoRa GW hex. ID conversion doubt

fleong, 2019-04-26:
Dear Pham,

I've read the doc "TTN and LoRaWAN". When registering the Gateway in TTN it is necessary to do a conversion in the hexadecimal ID. Could you clarify why 0000B827EBD1B236 becomes B827EBFFFFD1B236?

Best regards, 
Fernando

CongducPham, 2019-04-26:
It is not mandatory to insert FFFF the proposed way, but it is quite a well adopted convention. RAK gateways used FFFE for instance. We also have some "translater" that can use a low-cost multi-channel gw as hardware and inject data to our post-processing stage for instance and it is a convenient way to distinguish between the 2 modes on TTN.

The fact is post_status_processing_gw.py will insert these FFFF in order to call TTNstat.py for reporting gw's status to TTN. So if you wish to change the naming, change in post_status_processing_gw.py.

regards,

fleong, 2019-04-26:
Ok thanks!

newkton1, 2019-08-21:
Pham.  Thanks for all your work on this (I went trout fishing in Pau once—about 50 years ago—it was a nice city!).
I set up a test Lora gateway using your code and a RPI and RFM95W and it seems to be working in some way but I see a long string of receive errors.  Could you look at at the output from choosing the 0 option in ./cmd.sh and explain what might be causing the large numbers of Receive error 1 lines?
I posted the code at GIST:
https://gist.github.com/newkton1/b145b290aa10660954cf78d6f3d5e017

CongducPham, 2019-08-22:
Hi, do you start manually the lora_gateway with option 0 after boot? Because the lora_gateway program is already started at boot time. So looks like you have 2 instances of the gateway program running which will cause conflict on the SPI bus.

PS: where do you come from to had visited Pau 50 years ago?

newkton1, 2019-08-22:
Hello Pham.

Thanks for answering.  Yes, that was my mistake—there were two instances running.

I have another question if you don’t mind tech support!  When I run the Web admin I/F and set the frequency to 913.8 MHz and reboot the gateway, the change does not seem to be reflected in the ./cmd.sh screen when choosing option 5 (see screenshot). Is there something else I need to change?

I am originally English and our family was very Francophile, so my parent took me to France each summer where we spent long holidays trout fishing in the rivers running down from the Pyrenees, near Pau. It is a very beautiful spot.

After university, I came to Japan for work and have been here ever since (35 years now) after marrying and having children.  

One day I hope to get back to southern France for more trout fishing!

Best,

ROBERT HANCOCK
rockjapan@mac.com