CongducPham/LowCostLoRaGw #246: Compiler Warning: String Constant to Char*

arky, 2019-04-27:
LowCostLoRaGw/Arduino/Arduino_LoRa_Generic_Simple_MultiSensors/Arduino_LoRa_Generic_Simple_MultiSensors.ino:341:123: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   sensor_ptrs[0] = new HCSR04("DIS", IS_NOT_ANALOG, IS_CONNECTED, low_power_status, (uint8_t) 6, (uint8_t) 41, (uint8_t) 7); // for the MEGA
                                                                                                                           ^

https://github.com/CongducPham/LowCostLoRaGw/blob/b999ced249ac8c761eb0c97c4acc0aea7420e627/Arduino/Arduino_LoRa_Generic_Simple_MultiSensors/Arduino_LoRa_Generic_Simple_MultiSensors.ino#L341

Arduino IDE 1.8.9 
OS: $ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.10
Release:	18.10
Codename:	cosmic

CongducPham, 2019-04-27:
Thank you, I've solved these issues, will push new version on github.
regards,

arky, 2019-04-28:
That was quick! Thanks @CongducPham