CongducPham/LowCostLoRaGw #248: Pro-mini board and DIO pins

robshep, 2019-05-08:
Hi,  I got a batch of boards made and disappointed to find that only DIO0 is brought out to a pin.

I've had to bridge over with wire to get DIO1 connected so LMIC will run. 

It'd be great to see DIO1 and RESET brought out next to the D2 bridge too! 

THanks

Rob

CongducPham, 2019-05-08:
Yes, I known, I designed the board before testing LMIC. Right now I'm doing as you with a bridge wire. When I have time, will wire DIO1 and RST as well. Alternatively, use this board from a colleague of mine Fabien Ferrero: https://github.com/FabienFerrero/UCA_Board. Using 2, 7 and 9. regards,

robshep, 2019-05-08:
Thank you. And thanks for the board. 

Nevertheless
I have quite a few which I don’t need now. If you know anybody that would like one or two then get in touch to mail me a return envelope and i’ll post for free.

CongducPham, 2019-05-08:
Well, you also try our code that does not need any DIO. And it can send LoRaWAN packet to a LoRaWAN gateway as well. regards,

robshep, 2019-05-08:
Ok, thank you.
please could you highlight this example for me

CongducPham, 2019-05-08:
https://github.com/CongducPham/LowCostLoRaGw/tree/master/Arduino/Arduino_LoRa_temp
and
https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-aes_lorawan.md
https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-TTN.md

The advantage is that low power is completly handled, which is not the case for the current LMIC.

robshep, 2019-05-08:
Thank you.
Will try this now!

https://github.com/CongducPham/LowCostLoRaGw/tree/master/Arduino/Arduino_LoRa_temp

https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-aes_lorawan.md
https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-TTN.md

jamesee, 2019-05-08:
> Thank you. And thanks for the board.
> 
> Nevertheless
> I have quite a few which I don’t need now. If you know anybody that would like one or two then get in touch to mail me a return envelope and i’ll post for free.

Hi Robshep

I am looking for the Arduino Pro Mini and RFM95W breakout board. May I have 2 of your board?

How do I mail you a return envelope? I am based in Singapore.

Regards
James

robshep, 2019-05-08:
@jamesee DM me on twitter @95point2

jamesee, 2019-05-08:
> @jamesee DM me on twitter @95point2

Noted with thanks.

CongducPham, 2019-05-28:
Hi, I've made upgrade to the PCB to have DIO0, DIO1, DIO2 and RST if needed.

https://github.com/CongducPham/LowCostLoRaGw#pcbs

robshep, 2019-05-30:
Wow, this is great.  Thank you @CongducPham !

robshep, 2019-06-10:
Hi @CongducPham 

Thanks for adding these pins-connections.  Great.  Is the PCB source available for modification. I have a colleague who can design PCBs and we might want to add a couple of sensor footprints if you can opensource the design?

CongducPham, 2019-06-11:
Yes, I will put the .sch and .brd files as well.