CongducPham/LowCostLoRaGw #255: Lora no sent if define GPS_PIN_POWER

Alpha130, 2019-06-05:
In the program "Arduino_Lora_GPS"
If I comment GPS_PIN_POWER, the program works and sends every X minutes its GPS coordinates but the GPS does not go off completely.
If I unmount GPS_PIN_POWER to turn off the GPS, the data is no longer sent to the gateway

[Arduino_LoRa_GPS.zip](https://github.com/CongducPham/LowCostLoRaGw/files/3257698/Arduino_LoRa_GPS.zip)

Why ? 
Thanks and congratulations for your work

CongducPham, 2019-06-05:
How do you connect the GPS to the Arduino? Are you using the default pin 8 to power the GPS? Are you using a MOSFET transistor?

regards,

Alpha130, 2019-06-06:
GPS  <==> Arduino
RXD  ------  TX
TXD  ------  RX
GND => MOFSET 
VCC  ----- VCC

Yes, i use a MOFSET

[image]

Alpha130, 2019-06-06:
If I comment digitalWrite(GPS_PIN_POWER_PIN,LOW); it's done. I don't understand.
I try to add delay (2000) but it does not work

CongducPham, 2019-06-06:
So you mean the node does not send any message if you uncomment GPS_PIN_POWER?

Alpha130, 2019-06-06:
I do not see what's wrong

CongducPham, 2019-06-06:
Well, that's strange indeed. Check the MOSFET connection just in case.

Gate   ----------  Arduino's pin 8
Source ---------  Arduino's GND
Drain  ----------------------- ------ GPS's GND

Alpha130, 2019-06-06:
[image]

It's ok ? White => GPS's GND

CongducPham, 2019-06-06:
Looks OK. Unfortunately I don't see any obvious reason it is not sending when you uncomment GPS_PIN_POWER

Alpha130, 2019-06-06:
It'ok if i move 
`#ifdef GPS_PIN_POWER
    digitalWrite(GPS_PIN_POWER_PIN,LOW);
#endif `
after sending data

CongducPham, 2019-06-06:
OK, but really strange and I don't have the explanation.