CongducPham/LowCostLoRaGw #258: MQTT do not post to broker

viniaero, 2019-07-15:
do not post to broker

I see in the log, there is no error, but when I monitor using MQTT.fx nai nothing appears.

[image]

[image]

2019-07-15T11:11:37.891377> 
2019-07-15T11:11:37.892542> adding time zone info
2019-07-15T11:11:37.893917> new rcv timestamp (^t): 2019-07-15T11:11:37.8260
2019-07-15T11:11:37.895068> got first framing byte
2019-07-15T11:11:37.896227> --> got LoRa data prefix
2019-07-15T11:11:37.897396> --> DATA with_appkey: read app key sequence
2019-07-15T11:11:37.904113> app key is  0x05 0x06 0x07 0x08
2019-07-15T11:11:37.905358> app key disabled
2019-07-15T11:11:37.906535> valid app key: accept data
2019-07-15T11:11:37.907726> number of enabled clouds is 3
2019-07-15T11:11:37.919136> --> cloud[0]
2019-07-15T11:11:37.921895> uploading with python CloudThingSpeak.py
2019-07-15T11:11:37.923182> python CloudThingSpeak.py "#3#TC/28.22" "1,26,3,162,13,6,-43" "125,5,12,920360" "2019-07-15T11:11:37.8260" "000000E04C534458"
2019-07-15T11:11:50.682029> ThingSpeak: uploading (multiple)
2019-07-15T11:11:50.683444> rcv msg to log (\!) on ThingSpeak ( default , 3 ): 
2019-07-15T11:11:50.686593> ThingSpeak: will issue curl cmd
2019-07-15T11:11:50.691088> curl -s -k -X POST --data field3=28.22 https://api.thingspeak.com/update?key=XXXXXXXXXXXXXXXXXXX
2019-07-15T11:11:50.696753> ThingSpeak: returned code from server is 1138
2019-07-15T11:11:50.709058> --> cloud[1]
2019-07-15T11:11:50.710985> uploading with python CloudNodeRed.py
2019-07-15T11:11:50.712403> python CloudNodeRed.py "#3#TC/28.22" "1,26,3,162,13,6,-43" "125,5,12,920360" "2019-07-15T11:11:37.8260" "000000E04C534458"
2019-07-15T11:11:51.566757> CloudNodeRed: will generate json entry
2019-07-15T11:11:51.568046> {"source":"waziup/UPPA/Sensor3","measure":"TC","value":"28.22"}
2019-07-15T11:11:51.588200> --> cloud[2]
2019-07-15T11:11:51.590280> uploading with python CloudMQTT.py
2019-07-15T11:11:51.591749> python CloudMQTT.py "#3#TC/28.22" "1,26,3,162,13,6,-43" "125,5,12,920360" "2019-07-15T11:11:37.8260" "000000E04C534458"
2019-07-15T11:11:52.531079> --> cloud end

CongducPham, 2019-07-15:
Hi, I've seen this error and still working on it to see why it is not uploading. will get back to you.

regards,

viniaero, 2019-07-15:
ok.
Tks

CongducPham, 2019-07-22:
Do you still have this issue, seems that it is ok now.

viniaero, 2019-07-22:
Hello
for internal server (192.168.1.24 - in my case) is not working yet.

already with external server (mosquitto.org test), it is working:

Internal log

2019-07-22T17:12:19.690760> --> cloud[2]
2019-07-22T17:12:19.692003> uploading with python CloudMQTT.py
2019-07-22T17:12:19.693121> python CloudMQTT.py "#3#TC/28.81" "1,26,3,56,13,8,-56" "125,5,12,920360" "2019-07-22T17:12:15.9510" "000000E04C534458"
2019-07-22T17:12:20.639719> --> cloud end

External log:

2019-07-22T16:57:18.992924> --> cloud[2]
2019-07-22T16:57:18.994191> uploading with python CloudMQTT.py
2019-07-22T16:57:18.995404> python CloudMQTT.py "#3#TC/28.91" "1,26,3,54,13,6,-57" "125,5,12,920360" "2019-07-22T16:57:15.2680" "000000E04C534458"
2019-07-22T16:57:21.474637> MQTT: publishing
2019-07-22T16:57:21.475974> CloudMQTT: will issue cmd
2019-07-22T16:57:21.477161> mosquitto_pub -h test.mosquitto.org -t viniaero/vini/temp3/TC -m "28.91"
2019-07-22T16:57:21.481870> ['mosquitto_pub', '-h', 'test.mosquitto.org', '-t', 'viniaero/vini/temp3/TC', '-m', '28.91']
2019-07-22T16:57:21.484796> MQTT: publish success
2019-07-22T16:57:21.494140> --> cloud end