CongducPham/LowCostLoRaGw #266: How to connect gateway to the TTN, I tried, but, the status was not connected

irwan-syh, 2019-11-03:
How to connect gateway to the TTN, I tried, but, the status was not connected, i am use LoRa SX1278,
please explain, if u have idea

[image]

bkrajendra, 2019-12-24:
no success for me too.
Any idea how you configured gateway on ttn.

CongducPham, 2019-12-24:
Hi, did you check that you have the following setting?
- set ["radio_conf"]["mode"] to 11 in gateway_conf.json
- set ["gateway_conf"]["raw"] to true in gateway_conf.json
- set ["gateway_conf"]["aes_lorawan"] to false as encrypted data will be pushed to TTN
- check that python CloudTTN.py has been be added and enabled in the LoRaWAN encrypted cloud section of clouds.json:
"lorawan_encrypted_clouds" : [
 		{	
 			"name":"TheThingsNetwork cloud",
 			"script":"python CloudTTN.py",
 			"type":"TTN",			
 			"enabled":true			
 		},	
 		...	
 	]

CongducPham, 2019-12-24:
Also make sure that you have the correct gw id. I see on your screenshot that the gw id you are using for TTN is b827ebffffefc4a6, it means that the MAC address of your RPI must be b827ebefc4a6, is that correct?

ivanspirit, 2020-01-02:
Happy New Year to All !!
First I'll want to express my greetings to the great work you've done.

Sorry for the stupid question but I've tried a lot and read carefully everything and still no success to connect the GW to the TTN. Proper configuration for the json files and modes and AES ext. but where to put the long key from TTN-server provided the "Gateway Key"? May be I'm missing something very important here?!

bkrajendra, 2020-01-03:
me too. I was trying it to work with 433Mhz but nothing worked. tried lot other gateway implementation but nothing worked with RPI. Only ESP8266 gateway worked.

I ordered 868Mhz Modules now (RF95 finally) and will try with it as most supports are with 915 and 868Mhz frequencies.

Will update my findings here.

CongducPham, 2020-01-03:
To connect the GW to TTN, at the gateway side the lora mode should be set to 11, raw format should be set to true, aes_lorawan set to false and CloudTTN.py enabled in the "lorawan_encrypted_clouds" section of clouds.json. See here https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-TTN.md#ttn-and-lorawan

At the device, take the Arduino_LoRa_temp.ino example and uncomment the following define statements:
#define WITH_AES
#define LORAWAN
#define MY_FREQUENCY 868.1

then fill in the NetSKey, AppSKey and DevAddr with those from the TTN console.

///////////////////////////////////////////////////////////////////
//For TTN, use specify ABP mode on the TTN device setting

///////////////////////////////////////////////////////////////////
//ENTER HERE your App Session Key from the TTN device info (same order, i.e. msb)
**unsigned char AppSkey[16]** = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
///////////////////////////////////////////////////////////////////

//this is the default as LoRaWAN example
//unsigned char AppSkey[16] = { 0x2B, 0x7E, 0x15, 0x16, 0x28, 0xAE, 0xD2, 0xA6, 0xAB, 0xF7, 0x15, 0x88, 0x09, 0xCF, 0x4F, 0x3C };

///////////////////////////////////////////////////////////////////
//ENTER HERE your Network Session Key from the TTN device info (same order, i.e. msb)
**unsigned char NwkSkey[16]** = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
///////////////////////////////////////////////////////////////////

//this is the default as LoRaWAN example
//unsigned char NwkSkey[16] = { 0x2B, 0x7E, 0x15, 0x16, 0x28, 0xAE, 0xD2, 0xA6, 0xAB, 0xF7, 0x15, 0x88, 0x09, 0xCF, 0x4F, 0x3C };

///////////////////////////////////////////////////////////////////
// LORAWAN OR EXTENDED DEVICE ADDRESS FOR LORAWAN CLOUD
#if defined LORAWAN || defined EXTDEVADDR
///////////////////////////////////////////////////////////////////
//ENTER HERE your Device Address from the TTN device info (same order, i.e. msb). Example for 0x12345678
**unsigned char DevAddr[4]** = { 0x12, 0x34, 0x56, 0x78 };
///////////////////////////////////////////////////////////////////

ivanspirit, 2020-01-03:
Hi Congduc and thanks for the guide!

Now I'm able to connect the GW to TTN and the status is "Connected" (My mistake was that I was trying to put different MAC as a GW ID but didn't noticed that after reset the GW ID in json file was changed by the software) 

New problem arises then, after sending a packet from lorawan node the lora_gateway starting to give endless "Receive Error 1" code?
Any ideas ?

CongducPham, 2020-01-04:
Good to hear! An update of the web admin interface has been realized to better handle lorawan and also allow locally installed ChirpStack network server. best regards,