CongducPham/LowCostLoRaGw #268: Kernel panic error while booting the SDCARD on RPI0W

asprakash, 2020-01-20:
I tried latest [buster image](http://cpham.perso.univ-pau.fr/LORA/WAZIUP/raspberrypi-buster-WAZIUP-demo.iso.zip) on RPI Zero W. I am getting kernel panic error. I used class 10 sdcard with  Etcher tool. Tried with unzip method and directly flashed the zip file as well. 

TIA

CongducPham, 2020-01-21:
Hi, let me see I actually haven't tried on RPI0W.

sanbroz, 2020-09-19:
I am getting the same problem on Raspberry Pi 1 
[image]
B

hferee, 2020-10-18:
Same here on Rpi 1 as well.

CongducPham, 2020-10-19:
RPI1 & RPI0W have quite similar architecture so the issue probably comes from the same cause. Right now, these platforms are not in our priority line because they have limited processing power and we are moving into more functionalities at the edge of the IoT network. regards,

Samuel-ZDM, 2021-03-02:
Hi! 
So, it doesn't work on RaspI1? 
Same mistake here. 
Thanks.

CongducPham, 2021-03-02:
Well, I don't think that we will try to support RPI0 or RPI1. Given our development line towards more embedded AI, we are targeting RPI3 at least.