CongducPham/LowCostLoRaGw #270: no rx found

yannick-dev, 2020-02-15:
Hi, i tried to connect a mbed sx1272 to  a pi 3B over spi .
The pi is using the sd immage from the Waziup side.
I used the spi connection shown in the README and the one provided by mbed on the sx1272 side.
The webinterface shows a red message "no rx found".
My understanding was that any sx1272 should be compatible over SPI?
Is this likely to be a chip issue ?

Link to shield:
https://os.mbed.com/components/SX1272MB2xAS/
[image]
[image]

CongducPham, 2020-02-17:
Hi, yes, it should work. Do you have a device sending to the gateway? Otherwise, just run the low-level gateway ( sudo ./lora_gateway ) and see if the module is correctly detected. regards,

yannick-dev, 2020-02-17:
Thank you for your quick answer.

I updated the lora_gateway folder and rebuild it to be sure everything is up to date.
When i run ./lora_gateway the result is no transciever found. 
i have a node  running a otaa TTN request just next to the gateway, it times out.
What could i test to narrow down down the problem or would it be easier to get just a nother chip that is more reliable?

ppiwowar, 2020-02-18:
Another one like that here ...
RPi/3, 433MHz  radio (https://www.dfrobot.com/product-1664.html - I think SX127X), gateway updated to the recent version -> no rx found
INTERESTINGLY: after changing radio.makefile to:
            CFLAGS=-DPABOOST -DMAX_DBM=14 -DBAND433
and make lora_gateway
communication works like a charm (Arduino_LoRa_Ping_Pong, same radio = https://www.dfrobot.com/product-1664.htm, NodeMCU) BUT ONLY in cmd.sh after:
- K- kill all gateway related processes  and then
- 1- sudo ./lora_gateway --mode 1    
BUT web interface shows red message 'no rx found' AND sometimes it catches up after few minutes.
Configuration issue ? 
This gateway is really great so I look forward to have it fully working. Any suggestions, please?

CongducPham, 2020-02-18:
if you are using 433MHz band, then yes, -DBAND433 is needed if you want to run automatically the gateway on this band, otherwise, use web interface to indicate a frequency in 433MHz band.

It is normal that you have to kill the processes because when the gateway boot, al the software are started. Actually, after your modification, you just have to reboot.

The rx indication on the radio tab is only refreshed if you reload the page. Use the packet logger for real time.

regards,

CongducPham, 2020-02-21:
For TTN, you have to enable the LoRaWAN mode and the TTN clouds.

ppiwowar, 2020-02-21:
Thank You.
In fact, I was struggling with DGETOPT_ISSUE which needs special attention for Buster/linux configuration.
My gateway is fully operational now with radio https://www.dfrobot.com/product-1664.htm. Same radios with nodes. Playing with various clouds now and willing to add InfluxDB + Grafana this weekend.

CongducPham, 2020-02-21:
Good to hear!
regards,