CongducPham/LowCostLoRaGw #271: After  update in webapp cant   Run Basic config

dekpri, 2020-03-15:
i have Full update to version 502 
after that i have to Run Basic config
but it say "Date of last configuration undefined"
and gateway was fail 

please help

dekpri, 2020-03-15:
additional information 
it;s error in lora_gateway.cpp

`pi@raspberrypi:~/lora_gateway $ make lora_gateway_pi2
g++ -DPABOOST -DMAX_DBM=14 -DBAND868 -DRASPBERRY2 -c lora_gateway.cpp -o lora_gateway_pi2.o
lora_gateway.cpp: In function ‘int main(int, char**)’:
lora_gateway.cpp:1381:39: error: ‘optarg’ was not declared in this scope
            case 'a' : loraMode = atoi(optarg);
                                       ^~~~~~
lora_gateway.cpp:1381:39: note: suggested alternative:
In file included from /usr/include/arm-linux-gnueabihf/bits/getopt_posix.h:27,
                 from /usr/include/unistd.h:869,
                 from arduPi_pi2.h:248,
                 from SX1272.h:36,
                 from lora_gateway.cpp:197:
/usr/include/arm-linux-gnueabihf/bits/getopt_core.h:36:14: note:   ‘unistd::optarg’
 extern char *optarg;
              ^~~~~~
make: *** [makefile:21: lora_gateway_pi2.o] Error 1
`

CongducPham, 2020-03-15:
Are you using Buster? If yes, then uncomment in raspberry.makefile the line

dekpri, 2020-03-16:
It's Work!!

closed 2020-03-16 by dekpri

CongducPham, 2020-03-16:
great to hear. all the best.