CongducPham/LowCostLoRaGw #273: Mode 11 cannot receive data from Arduino_LoRa_temp

ktannady, 2020-03-17:
Hello i have successfully connected my gateway to my local chirpstack and TTN,
i also can get my Rpi gateway to play ping pong with my end node (with mode 1).
now that I change to mode 11 to connect to my chirpstack, i cannot get my data from the Arduino_LoRa_temp ( i can receive data if i set my gateway to mode 1).
i have uncommented `#WITH_AES` and `#define LORAWAN` but still no success

Here is my code on the Arduino
`/*
 *  temperature sensor on analog A0 to test the LoRa gateway
 *  extended version with AES and custom Carrier Sense features
 *  
 *  Copyright (C) 2016-2019 Congduc Pham, University of Pau, France
 *
 *  This program is free software: you can redistribute it and/or modify
 *  it under the terms of the GNU General Public License as published by
 *  the Free Software Foundation, either version 3 of the License, or
 *  (at your option) any later version.

 *  This program is distributed in the hope that it will be useful,
 *  but WITHOUT ANY WARRANTY; without even the implied warranty of
 *  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *  GNU General Public License for more details.
 *
 *  You should have received a copy of the GNU General Public License
 *  along with the program.  If not, see <http://www.gnu.org/licenses/>.
 *
 *****************************************************************************
 * last update: Feb 20th, 2020 by C. Pham
 * 
 * This version uses the same structure than the Arduino_LoRa_Demo_Sensor where
 * the sensor-related code is in a separate file
 */

///////////////////////////////////////////////////////////////////////////////////////////////////////////
// add here some specific board define statements if you want to implement user-defined specific settings
// A/ LoRa radio node from IoTMCU: https://www.tindie.com/products/IOTMCU/lora-radio-node-v10/
//#define IOTMCU_LORA_RADIO_NODE
// uncomment if you are sure you have an ESP8266 board and that it is not detected
//#define ESP8266
///////////////////////////////////////////////////////////////////////////////////////////////////////////

#include <SPI.h> 
// Include the SX1272
#include "SX1272.h"
#include "my_temp_sensor_code.h"

/********************************************************************
 _____              __ _                       _   _             
/  __ \            / _(_)                     | | (_)            
| /  \/ ___  _ __ | |_ _  __ _ _   _ _ __ __ _| |_ _  ___  _ __  
| |    / _ \| '_ \|  _| |/ _` | | | | '__/ _` | __| |/ _ \| '_ \ 
| \__/\ (_) | | | | | | | (_| | |_| | | | (_| | |_| | (_) | | | |
 \____/\___/|_| |_|_| |_|\__, |\__,_|_|  \__,_|\__|_|\___/|_| |_|
                          __/ |                                  
                         |___/                                   
********************************************************************/

///////////////////////////////////////////////////////////////////
// COMMENT OR UNCOMMENT TO CHANGE FEATURES. 
// ONLY IF YOU KNOW WHAT YOU ARE DOING!!! OTHERWISE LEAVE AS IT IS
// 
// FOR UPLINK WITH ONLY AES ENCRYTION: uncomment WITH_AES
// FOR UPLINK TO LORAWAN CLOUD ENCAPSULATED LORAWAN FORMAT: uncomment WITH_AES & EXTDEVADDR 
// FOR UPLINK TO LORAWAN CLOUD NATIVE LORAWAN FORMAT: uncomment LORAWAN
// FOR UPLINK TO LORAWAN CLOUD AND DOWNLINK WITH NATIVE LORAWAN FORMAT: uncomment LORAWAN & WITH_RCVW
// 
// more info: https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-aes_lorawan.md
//
// DEFAULT CONFIGURATION: LoRaMode 1 (SF12BW125), no encryption, no LoRaWAN, no downlink
//
//#define WITH_EEPROM
[... 1158 lines cut ...]
15:31:39.628 -> transmitted LoRaWAN-like packet:
15:31:39.662 -> MHDR[1] | DevAddr[4] | FCtrl[1] | FCnt[2] | FPort[1] | EncryptedPayload | MIC[4]
15:31:39.662 -> 40 4A 1F 01 26 00 00 00 01 79 F8 50 C9 3B 02 32 D7 18 39 F8 EC 41 8F 
15:31:39.696 -> [base64 LoRaWAN HEADER+CIPHER+MIC]:QEofASYAAAABefhQyTsCMtcYOfjsQY8=
15:31:39.696 -> --> CS1
15:31:40.238 -> --> CAD 551
15:31:40.238 -> OK1
15:31:40.238 -> --> RSSI -127
15:31:41.731 -> End send: 7768
15:31:41.731 -> LoRa pkt size 23
15:31:41.731 -> LoRa pkt seq 0
15:31:41.765 -> LoRa Sent in 1489
15:31:41.765 -> LoRa Sent w/CAD in 2042
15:31:41.765 -> Packet sent, state 0
15:31:41.765 -> 0
15:31:41.765 -> Will send next value at
15:31:41.765 -> 607788
`
Here is the output on my rebooted Rpi
[image]

After 10 minutes
[image]

i have been stuck for days, thankyou.

CongducPham, 2020-03-17:
Looks like it comes from a mismatch frequency. In mode 1, the default frequency in band 433 is 433.3 and in LoRaWAN mode it is 433.175. Your gw is on 433.3 while your device is on 433.175. Do you use the latest version of lora_gateway.cpp as normally the gateway should be in 433.175 if it is compiled with BAND433, or did you set the 433.3 freq right in the gateway_conf.json? Also do you use the quick lorawan configuration feature of the web interface on the gateway? It is safer in order to able to set all parameters.

regards,

ktannady, 2020-03-17:
hello, i have tried to update as you do in your video, but i cannot run "sudo ./lora_gateway" anymore as it said that "lora_gateway command not found" so i don't try to update the gateway after that.
[image]

i tried to compile the lora_gateway.cpp to change the band but i couldn't because of some error (i forgot to screenshot it), so i change the band on the web interface to 433mhz.

CongducPham, 2020-03-17:
In radio.makefile, add the option for band 433:

CFLAGS=-DPABOOST -DMAX_DBM=14 -DBAND433

check whether you need PABOOST or not

Then, you should use the scripts/basic_config_gw.sh script, just run:

cd scripts
sudo ./basic_config.gw.sh

to setup and compile your gateway.

Read instructions on the github for updates, it is better to use the web interface.

regards,

ktannady, 2020-03-18:
Hello, after run the basic config either from the bash or the web interface, i still cannot run my lora_gateway command.

[image]

Here is my radio.makefile as you suggest
[image]

regards,

CongducPham, 2020-03-18:
If you have Buster version of Raspbian, then be sure in raspberry.makefile to have uncommented the following line:

CFLAGS+=-DGETOPT_ISSUE

ktannady, 2020-03-18:
Hello, i have successfully run my ./lora_gateway now. I have still the same issues that my gateway cannot receive data from my arduino side.

Here is my gateway setting on my web interface.
[image]

Here is my arduino side
[image]

Here is the result on the gateway side
[image]

is there anyway i can see which frequency my gateway is currently running?
regards,

ktannady, 2020-03-18:
Update : I have successfully receive data on my gateway after a few tries. it seems like there is some hardware (loose cable) that stop them from communicating,

Thankyou for helping me sir,
regards,

CongducPham, 2020-03-18:
Your gateway is still running on 433.3 while your device is running on 433.175 which is default freq for LoRaWAN. To use default frequency on gateway for LoRaWAN, you have to leave the frequency configuration on the gateway to -1 and not to 433.3.

ktannady, 2020-03-18:
im sorry sir, but i cant seem to find a -1 option on the other side except the mode one.

[image]

now im confused why i can still receive my data on these config.

regards,

ktannady, 2020-03-18:
alright, i have been successfully receive data from my arduino, i need to manually change the frequency to -1 in gateway_conf.json as the web interface doesnt have the option to change to -1.

Thankyou sir,
regards,

CongducPham, 2020-03-18:
You have to update the web interface as well. The latest version allows you to set freq to -1 in the ISM band option. Anyway, great to hear that it is working now.

ktannady, 2020-04-01:
Hello sir, does the gateway can handle chirpstack downlink?

CongducPham, 2020-04-01:
yes. please look at https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-downlink.md

ktannady, 2020-04-08:
Hello sir, after enqueue the downlink payload from my local chirpstack i cant seem to find the downlink.txt, is there anything that i missed out?

regards.

CongducPham, 2020-04-08:
You have to make sure that everything is configured for lorawan with downlink, so better to use the "configure for LoRaWAN" quick button on the web admin interface.

Then, downlink.txt will be created when PULL DATA will be received from network server, after the downlink has been processed, the file is deleted.

You can check if the gateway receives the downlink messages by looking at the log file. Output similar to:
	-----------------------------------------------------
	Check for downlink requests 2020-01-21T17:33:28.439246
	Read downlink: 204
	Downlink entry: {"txpk":{"imme":false,"rfch":0,"powe":14,"ant":0,"brd":0,"tmst":4156747970,"freq":868.1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":25,"data":"YCQfBCYC9QAIAQF+eOi0BvnZlM6hkIt/wA=="}}
	Process downlink request
	LoRaWAN downlink request
	Target RX1
	Packet sent, state 0
	-----------------------------------------------------

regards,

ktannady, 2020-05-15:
Hello sir, is there anyway i can make the gateway generate a downlink, without the slave communicating first?
i want to make the gateway send command via downlink every hour to the slave,

regards,

CongducPham, 2020-05-15:
Hi, there is no such thing possible in so-called class A devices. What you ask for is for class B devices where peridic downlink can be programmed. LMIC has normally support for class B and ChirpStack Network Server has also this support. You might go for that and test how it works.

hope that helps