CongducPham/LowCostLoRaGw #274: Change ChirpStack Server not possible

sheggy012, 2020-03-25:
Hello together I have dedicated Chirpstack server running in a VM, can I have my gateway sent to this address? In the admin interface there is an option to set a Chirpstack IP, but the submit button does not work. If I change the address in the CloudChirpStack.py it seems to be not working either. Gateway is still shown as offline.

sheggy012, 2020-03-26:
Okay, found the problem. It seems that the scripts/lorawan_stats/ttn_stats.py and the cs_stats.py cannot read the key_TTN.py and key_Chirpstack.py The value under 
```
except ImportError:
        lorawan_server="127.0.0.1"
```

is always taken.

CongducPham, 2020-03-26:
Hi, first, verify that your configuration file (gateway_conf.json and the key files) follows the lastest version because to preserve existing configuration these file are not updated. So many your key_Chirpstack.py file is not up to date with the correct fields. Check the latest version on github and modify your file accordingly. I believe the issue comes from this problem. Then, when you change the ChirpStack server address, use the Configure for LoRaWAN quick button to also set the downlink server in gateway_conf.json to the correct lorawan server address.

sheggy012, 2020-03-26:
It's a fresh installation, all files are up to date. The problem is that the script scripts/lorawan_stats/cs_stats.py looks only in the same directory for key_Chirpstack.py but its 2 folder higher. You can easy test it. Just change the IP Adress in cs_stats.py in the section "except ImportError:" and it will get instantly used in the next status update.

CongducPham, 2020-03-26:
Hi, the script is called by post_status_processing_gw.py which is called by post_processing_gw.py using the script path so the current folder is still /home/pi/lora_gateway. I believe that the issue here is that you modified the server address but didn't reboot. Actually maybe I should add in cs_stats.py the line "sys.dont_write_bytecode = True" as it is the case in Cloud* .py file to allow dynamic changes of key_Chirpstack.py.

CongducPham, 2020-03-26:
Yes, you are right, even though the current directory is still /home/pi/lora_gateway, the import statement seems to fail. So you can just add

sys.path.insert(1, '/home/pi/lora_gateway')

right after import sys in cs_stats.py

thanks for raising this issue.
regards,

sheggy012, 2020-03-26:
With the change it works now.
Thanks for your support and your hard work.
Best Regards

closed 2020-03-26 by sheggy012